almindor/mipidsi #3: Optional RST handling

pyaillet, 2022-02-24:
Hi !

I was able to use this driver for a TTGO Display with st7789. However there is no RST pin on the TTGO T-Watch.
I would like to propose some changes to this driver to handle the absence of the RST Pin.

Would you prefer handling this by wrapping the RST Pin in an Option ? Creating different models ? Another approach ?

almindor, 2022-02-24:
Good question. I wonder if this should be handled by the `DisplayInterface` instead. That'd be a big change tho, for now I think we should make it optional somehow.

The issue is using just `Option<RST>` doesn't solve it because it still requires the type to be `OutputPin` of some kind. So even if you're passing `None` you'd have to specify the type. We could possibly add a constructor that uses some sort of dummy `OutputPin` type internally so users can avoid that issue I guess.

pyaillet, 2022-02-28:
Let's close this one two as the PR was merged :)

Thanks again !

closed 2022-02-28 by pyaillet