almindor/mipidsi #6: common_init function never called

brianmay, 2022-03-17:
When calling this as per instructions:

```
// create a DisplayInterface from SPI and DC pin, with no manual CS control
let di = SPIInterfaceNoCS::new(spi, dc);
// create the ILI9486 display driver in rgb666 color mode from the display interface and RST pin
let mut display = Display::ili9486_rgb666(di, rst);
// clear the display to black
display.clear(Rgb666::BLACK)?;
```

The `common_init` function is never called, and the screen is not initialised.

brianmay, 2022-03-19:
Will leave this for now, until some of the other bugs can get fixed.

I suspect that the `ili9486_rgb666` will need a new parameter for the delay struct, so it can call the init function of the model.

almindor, 2022-03-23:
Sorry maybe I'm being obtuse but what `common_init` function?

brianmay, 2022-03-23:
Sorry, got the name wrong:

https://github.com/almindor/mipidsi/blob/1b7d8e2d36596327bcc3867860c3558d47b6990e/src/models/ili9486.rs#L155-L181

It is called by the `init()` function here:

https://github.com/almindor/mipidsi/blob/1b7d8e2d36596327bcc3867860c3558d47b6990e/src/models/ili9486.rs#L89

But there is nothing that calls the `init()` function.

The `new` function is called, but the `new` function does nothing.

I just saw this method:

https://github.com/almindor/mipidsi/blob/1b7d8e2d36596327bcc3867860c3558d47b6990e/src/lib.rs#L136-L138

Maybe that calling this is intended. If so, the documentation on the `README.md` probably needs updating.

almindor, 2022-03-23:
Ah sorry you're right. `init_common` is meant to be a common setup for all ILI9486 displays in any configuration. The individual init functions are then used to setup specifics for e.g. different color variants.

PR #12 (2022-03-28): Add call to `init` in README.md
Fixes #6.
commit aa3b2070 (2022-03-28): Add call to `init` in README.md

Fixes #6.
closed 2022-03-28 by almindor