almindor/mipidsi #7: screen orientations are hardcoded

brianmay, 2022-03-17:
Some screens require different madctl 0x26 settings, and e.g. will appear reversed without.

For a possible solution see: https://github.com/yuri91/ili9341-rs/commit/8a4aee95c2934cd6aaac4b243ab8bda24d0b7475

brianmay, 2022-03-19:
This would be my recommended fix:

```diff
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -5,7 +5,7 @@
 //! This crate provides a generic ddisplay driver to connect to TFT displays
 //! that implement the [MIPI DSI](https://www.mipi.org/specifications/dsi).
 //! Currently only supports SPI with DC pin setups via the [display_interface]
-//!  
+//!
 //! An optional batching of draws is supported via the `batch` feature (default on)
 //!
 //! ## Example
@@ -64,8 +64,6 @@ where
 pub enum Orientation {
     Portrait = 0b0000_0000,         // no inverting
     Landscape = 0b0110_0000,        // invert column and page/column order
-    PortraitSwapped = 0b1100_0000,  // invert page and column order
-    LandscapeSwapped = 0b1010_0000, // invert page and page/column order
 }
 
 impl Default for Orientation {
@@ -147,9 +145,12 @@ where
     ///
     /// Sets display [Orientation]
     ///
-    pub fn set_orientation(&mut self, orientation: Orientation) -> Result<(), Error<RST::Error>> {
+    pub fn set_orientation(&mut self, orientation: Orientation, invertx: bool, inverty: bool) -> Result<(), Error<RST::Error>> {
         self.write_command(Instruction::MADCTL)?;
-        self.write_data(&[orientation as u8])?;
+        let mut data = orientation as u8;
+        if invertx { data = data ^ 0x40 };
+        if inverty { data = data ^ 0x80 };
+        self.write_data(&[data])?;
         self.orientation = orientation;
         Ok(())
     }
```

If this is acceptable I can do a PR.

almindor, 2022-03-28:
Could we close this one in favor of #8  please? (or the other way around)

brianmay, 2022-03-28:
Doesn't bother me if we do. I considered them distinct issues when I opened the reports, but there was considerable overlap so I used one PR to solve both of them.

brianmay, 2022-03-31:
I think this was accidentally closed :-)

PR #14 (2022-03-31): keep MADCTL value to use with orientation changes
Keeps MADCTL value from model `init` and uses it with orientation writes to keep the color and other definitions.

Should fix #7 and #8 and enable a fix for #10

I've also switched to pure static dispatch for all cases with `WriteOnlyDataCommand`
PR #9 (2022-04-04): Fix size and orientation issues
Fixes #7
Fixes #8
commit 98e64a62 (2022-04-04): Fix size and orientation issues

Fixes #7
Fixes #8