almindor/mipidsi #47: Document changed default color order

rfuest, 2022-12-15:
The default color order for some display was changed in #36, but this hasn't been documented in the changelog or the migration notes. 

This change will probably affect many users, so that it would also be a good idea to add a troubleshooting section for common color problems to the docs. Some example images and/or directions how to identify RGB/BGR or color inversion on TN vs IPS displays would be very useful IMO.

almindor, 2022-12-15:
Hmm, I thought I kept it what the original inits were using. Did I miss something?

almindor, 2022-12-16:
> You did remove the incorrect RGB/BGR inversion in this commit: [6a16384](https://github.com/almindor/mipidsi/commit/6a1638411d1f0536243bd21e34b4675346da5876)

Ah the 7789 yes. Must've been my display's setup matched it. I'll make it none are inverted by default in the next release.

rfuest, 2022-12-16:
> Ah the 7789 yes. Must've been my display's setup matched it. I'll make it none are inverted by default in the next release.

The color inversion for the ST7789 in the `default_options` is actually part of #48 and not this issue. This issue was meant to discuss the inverted RGB/BGR bit in `MADCTL` for ST7735s and ILI9342C.

almindor, 2022-12-16:
> The color inversion for the ST7789 in the `default_options` is actually part of #48 and not this issue. This issue was meant to discuss the inverted RGB/BGR bit in `MADCTL` for ST7735s and ILI9342C.

Aah right. In either case I think the standard should be no-inversion for any of the "defaults" regardless of model, unless there's one that for some reason is always wire specific.

almindor, 2024-01-30:
I believe this is fixed

closed 2024-01-30 by almindor