almindor/mipidsi #48: Remove color inversion form `default_options`

rfuest, 2022-12-15:
The `default_options` for some models contain `with_invert_colors(true)` and others don't. IMO it would be better to be consistent across all models and default to non inverted colors.

The decision if color inversion is necessary is entirely dependent on the LCD glass and not the controller model and can only be decided for a specific LCD module. For LCD module specific constructors, like the `st7789_pico1`, the `with_invert_colors(true)` should stay. To help users who want to use other LCD modules there should be additional docs to help them identify problems, as also mentioned in #47.

almindor, 2022-12-15:
Good call, I'll make sure to include in the next api breaking release.

rfuest, 2024-01-26:
The color inversion has been removed from the default options in #80.

closed 2024-01-26 by rfuest