almindor/mipidsi #49: Colors are reversed on st7735s tft display

togisoft, 2022-12-26:
I tried a lot, but I couldn't find a solution about the colors.

Rgb565::WHITE -> Outputs black
Rgb565::BLACK -> Outputs white

My code:
```Rust
    let spi = spi::Spi::new(
        peripherals.SPI2,
        io.pins.gpio4,   
        io.pins.gpio5,   
        io.pins.gpio25,   
        io.pins.gpio13,   
        100u32.kHz(),
        SpiMode::Mode0, 
        &mut system.peripheral_clock_control,
        &clocks,
    );

    let reset = io.pins.gpio26.into_push_pull_output();
    let di = SPIInterfaceNoCS::new(spi, io.pins.gpio15.into_push_pull_output());

    let mut display = mipidsi::Builder::st7735s(di)
        .with_display_size(128,160)
        .with_color_order(ColorOrder::Rgb) // Even if I did BGR the problem didn't go away
        .init(&mut delay, Some(reset))
        .unwrap();
    display.clear(Rgb565::BLACK).unwrap();

```

rfuest, 2022-12-27:
> I tried a lot, but I couldn't find a solution about the colors.

Did you already try to add `with_invert_colors(true)` to the builder calls?

togisoft, 2022-12-27:
Method not found

```Rust
  .with_invert_colors(true)
   |          ^^^^^^^^^^^^^^^^^^ method not found in `Builder<SPIInterfaceNoCS<Spi<esp32_hal::esp32::SPI2>, GpioPin<Output<PushPull>, Bank0GpioRegisterAccess, InputOutputAnalogPinType, 15>>, ST7735s>`
```

rfuest, 2022-12-27:
OK, sorry I did forget that this method wasn't included in the latest released version. You could try to use the GIT version until the next version is released.

togisoft, 2022-12-27:
```Rust
        .with_invert_colors(false)
```
run when false. thank you

almindor, 2023-06-11:
I believe this is outdated/fixed now, please re-open if this is still a problem.

closed 2023-06-11 by almindor