almindor/mipidsi #51: Improve readme example code

alyti, 2023-01-20:
Based on this [article](https://lilymara.xyz/posts/images-esp32/) that covered use of this library the following changes from it's summary could be applied to the current readme:

* mipidsi documentation should include required use statements in its example code, as well as linking out to required crates
* mipidsi documentation should not reference the no-longer-used Rgb666 color type

These changes would improve beginner journey.

almindor, 2023-01-21:
I can have a look, one problem is that a "real" example  case would need to use some sort of specific embedded-hal implementation to get the SPI for example. That would bring a whole tree of dependencies in, and would be generally useless as an example because it'd only work for people with that setup anyhow.

almindor, 2023-01-23:
> * mipidsi documentation should not reference the no-longer-used Rgb666 color type

What do you mean with this exactly? Rgb666 is a valid color, it's just RawU18 that will go away AFAIK