almindor/mipidsi #55: ST7789 display is black with mipidsi crate, until program execution is interrupted- working fine with st7789 crate

tingox, 2023-05-09:
This example
```
// nrf52840dk-mipidsi_examples - xtext
//
#![no_std]
#![no_main]

extern crate cortex_m_rt as rt;

extern crate nrf52840_hal;
extern crate panic_halt;

use cortex_m_rt::entry;
use defmt_rtt as _; // global logger
use display_interface_spi::SPIInterfaceNoCS;
use embedded_graphics::{
    mono_font::{ascii::{FONT_6X10, FONT_10X20}, MonoTextStyle},
    pixelcolor::Rgb565,
    prelude::*,
    text::Text,
};

use nrf52840_hal::gpio::p0::Parts;
use nrf52840_hal::gpio::Level;
use nrf52840_hal::prelude::*;
use nrf52840_hal::spim;
use nrf52840_hal::Delay;

use mipidsi::{Builder, Orientation};

#[entry]
fn main() -> ! {
    let core = nrf52840_hal::pac::CorePeripherals::take().unwrap();
    let mut delay = Delay::new(core.SYST);

    let p = nrf52840_hal::pac::Peripherals::take().unwrap();
    let port0 = Parts::new(p.P0);

    let _backlight = port0.p0_31.into_push_pull_output(Level::Low); // set medium backlight on
    // let rst = port0.p0_29.into_push_pull_output(Level::Low); // reset pin
    let rst = port0.p0_29.into_push_pull_output(Level::High);   // reset pin
    let _cs = port0.p0_28.into_push_pull_output(Level::Low);    // keep low while driving display
    let dc = port0.p0_30.into_push_pull_output(Level::Low);     // data/clock switch

    let spiclk = port0.p0_03.into_push_pull_output(Level::Low).degrade(); // SPI clock to LCD
    let spimosi = port0.p0_04.into_push_pull_output(Level::Low).degrade(); // SPI MOSI to LCD

    let pins = spim::Pins {
        sck: spiclk,
        miso: None,
        mosi: Some(spimosi),
    };

    // create SPI interface
    let spi = spim::Spim::new(p.SPIM0, pins, spim::Frequency::M8, spim::MODE_3, 122);

    // display interface abstraction from SPI and DC
    let di = SPIInterfaceNoCS::new(spi, dc);

    // create driver
    let mut display = Builder::st7789(di)          // known model
            .with_display_size(240, 240)           // set options
            .init(&mut delay, Some(rst)).unwrap(); // with reset pin

    display.set_orientation(Orientation::Portrait(false)).unwrap();

    // color constants:BLACK, WHITE, RED, GREEN, BLUE, CYAN, MAGENTA, YELLOW

    // make text style
    // Create a new character style
    let style_white = MonoTextStyle::new(&FONT_6X10, Rgb565::WHITE);
    let style_blue = MonoTextStyle::new(&FONT_10X20, Rgb565::BLUE);
    let style_red = MonoTextStyle::new(&FONT_10X20, Rgb565::RED);
    let style_yellow = MonoTextStyle::new(&FONT_10X20, Rgb565::YELLOW);

    // Create a text at position (20, 30) with the previously defined style
    let text1 = Text::new("Hello Rust!", Point::new(20, 30), style_white);

    // draw text on black background
    display.clear(Rgb565::BLACK).unwrap();
    text1.draw(&mut display).unwrap();
    
    delay.delay_ms(1000u16);
    // clear display to a white background
    display.clear(Rgb565::WHITE).unwrap();
    // write text in different colors
    Text::new("Different", Point::new(20,30), style_blue).draw(&mut display).unwrap();
    Text::new("colors", Point::new(20,50), style_red).draw(&mut display).unwrap();

    defmt::info!("Rendering done");

    loop {
        Text::new("test", Point::new(20,70), style_blue).draw(&mut display).unwrap();
        delay.delay_ms(250u8);
        Text::new("test", Point::new(20,70), style_yellow).draw(&mut display).unwrap();
        delay.delay_ms(250u8);
    }
}
```
does not produce  output on the display until the program is interrupted. If I use the st7789 crate instead, it works without problems.
Is there a way to turn off the batch feature (I couldn't figure out a way from the documentation) so I can test if that has anything to do with the problem?

almindor, 2023-06-11:
Yes you can disable it by using `--no-default-features` with `cargo build --no-default-features`.

Sorry about the delay, I somehow missed this issue.

rfuest, 2023-06-11:
I believe this is a duplicate of #56 and can be closed.

closed 2023-08-05 by almindor