almindor/mipidsi #57: mipi dsi OR spi?

x931890193, 2023-05-11:
Is it the mipi dsi protocol or the spi protocol？

almindor, 2023-05-12:
It's a MIPI DSI "generic" driver. Currently only SPI is supported by the `display-interface` crate used by it though (AFAIK).

closed 2023-05-12 by almindor

almindor, 2023-05-12:
> > It's a MIPI DSI "generic" driver.
> 
> Is that correct? I wanted to bring this up before, because the crate's readme is misleading IMO. The first sentence of the readme states that "This crate provides a generic display driver to connect to TFT displays that implement the MIPI DSI." But this crate implements display drivers for displays which use the MIPI DCS and not DSI. While DSI displays use the DSC command set there is no support for the actual DSI protocol in this crate.

Yes correct my bad. The name is a bit unfortunate. Naming it MIPIDCS wouldn't be better though because that'd imply it's the DCS command set crate...

I'm up for a rename if a really good fitting name can be found.

> 
> > Currently only SPI is supported by the `display-interface` crate used by it though (AFAIK).
> 
> As far as I know there are currently no implementations of `display-interface` for MIPI DSI, but I've at least used a parallel interface before.

Yes sorry. You are correct. What I meant was that SPI was the only transport in DI but I see it has i2c and parallel as well now.

rfuest, 2023-05-12:
> Yes correct my bad. The name is a bit unfortunate. Naming it MIPIDCS wouldn't be better though because that'd imply it's the DCS command set crate...
> 
> I'm up for a rename if a really good fitting name can be found.

I don't have a good idea for a new name. Until a new name might be found I would suggest to just improve the documentation to make sure the purpose of the crate is obvious.

Adding the crate to the e-g display driver list at https://github.com/embedded-graphics/embedded-graphics#display-drivers would also make it more discoverable for e-g users, regardless of the name.