almindor/mipidsi #62: Is it still true that the ST7789 only works in SPI? IF so what needs to be done to make it work on 8bit parallel

jscatena88, 2023-06-10:
This comment indicates it only works with SPI mode: https://github.com/almindor/mipidsi/blob/321daa0195968de62ca187fb01305e26dac0b454/src/models/st7789.rs#L21

Is that still the case? I was able to construct an instance using a `PGPIO8BitInterface`, but it is not working properly. I'm not sure if that is expected or not at this point

rfuest, 2023-06-11:
The comment is outdated. I know at least one project that is using an ST7789 display with `mipidsi` on an parallel bus: https://github.com/nw-rs/board-support/blob/master/src/display.rs That project uses a special hardware peripheral inside an STM32 instead of `PGPIO8BitInterface`, but it should work the same.

Did you not get any display at all or was the display corrupted?

jscatena88, 2023-06-11:
I actually ended up getting it to work. I was just fumbling and forgot ChipSelect is active low not active high. But I can now confirm that the ST7789 works with PGPIO8BitInterface. Specifically I am using the Lilygo T-HMI

rfuest, 2023-06-11:
OK, great to hear that it works. And now that you have mentioned the T-HMI I also remembered that I've successfully tried `mipidsi` on very similar hardware in the past, a T-DISPLAY S3.

jscatena88, 2023-06-11:
I'll go ahead and close this issue but it's probably worth pulling out the comment, especially since it shows up in the docs on docs.rs as well. I almost didn't even give this library a shot because of it

closed 2023-06-11 by jscatena88