almindor/mipidsi #67: Final part of GC9A01 display cut off

Skilvingr, 2023-07-12:
Hi, I have a 240x240 GC9A01. This is the code I'm using to test the display (I'm using the latest git version of the crate):
```
buffer_provider.display.clear(Rgb565::WHITE); // Start with a clean display

delay.delay_ms(1000u32);

// Try to fill the display line after line
for i in 0u16..240 {
    buffer_provider.display.set_pixels(0u16, i, 240u16, i, vec![Rgb565::BLACK; 240]); // This cuts
}

delay.delay_ms(1000u32);

// Try to fill the display pixel after pixel
for i in 0u16..240 {
    for j in 0u16..240 {
        buffer_provider.display.set_pixels(j, i, j, i, vec![Rgb565::RED]); // This doesn't cut
    }
}
```
`clear` works as expected, same for the third try (pixel after pixel), but the second try (the one with the range) cuts off the last part of the display:

[image]

Does anyone know why this happens? Thanks!

rfuest, 2023-07-12:
Not sure if that's the problem, but in the line by line example the third parameter should be `239u16`, because the coordinates are in the range `0..=239`.

Skilvingr, 2023-07-12:
Great! That was the problem. I thought that the upper part of the range would have been taken into account by the crate itself.
Thank you very much indeed!

closed 2023-07-12 by Skilvingr

rfuest, 2023-07-12:
> I thought that the upper part of the range would have been taken into account by the crate itself.

`set_pixels` doesn't, because it's a low level function, which is mainly intended to be used by other frameworks. We have just recently updated the docs a few days ago to emphasize this (#66).

If you use `embedded-graphics` to draw something to the display it will be automatically clipped and you won't get any artifacts from trying to draw outside the display area.