almindor/mipidsi #68: Add support for 12 bit color (RGB444)

cosmikwolf, 2023-07-21:
This issue is dependent on RGB444 and BRG444 being added to embedded-graphics pixelcolor, but I figured it would be good to create this bug to track the issue as well. 

Adding 12 bit color support would result in a 25% decrease in the time it takes to write color data to a display compared to 16 bit colors, utilizing the RGB444 mode as exists on chips like the ST7789

https://github.com/embedded-graphics/embedded-graphics/issues/726

pferreir, 2025-05-18:
I think Rgb444 is already available.

rfuest, 2025-05-18:
`Rgb444` is not yet available in a stable release of e-g, but it would be nice to have the option to use it if color resolution isn't important. 

`Rgb332` was also added to e-g and I know I've seen it used in some display controller datasheet recently, but I'm not sure if it was a type `mipidsi` already supports. `Rgb111` with two pixels per byte (1/4 the bandwidth requirement of `Rgb565`) would also be worth investigating, which at least some `ILI...` controllers support.