almindor/mipidsi #73: init without a RST pin

MathyV, 2023-09-09:
Hi,

I almost feel stupid having to ask this but I can't seem to figure it out on my own: I have a device where the ST7789 doesn't have a separate RST pin, it is connected to the RST pin of the ESP32. I got it working with the driver, which is great, but I can't seem to run the `init` function without passing a real Pin to it.

Obviously I don't want to use a fake pin, in the first place not because there are actually no free pins left on the device so I can't use other functionality.

In one of the examples it says `None::<OutputPin>`, which doesn't work. I also tried with `None::<AnyOutputPin>` and many other variants but it always comes down to either errors that the size isn't known at compile time and/or the trait `embedded_hal::digital::v1::OutputPin` is not implemented.

Could it be related to the `v1` vs `v2` implementation of digital io in `embedded-hal` vs `esp-idf-hal`? Or am just being stupid and am I failing to see the obvious here? Either way I'd appreciate your input :rofl: 

Thx,
Mathy

almindor, 2023-09-10:
> In one of the examples it says `None::<OutputPin>`, which doesn't work. I also tried with `None::<AnyOutputPin>` and many other variants but it always comes down to either errors that the size isn't known at compile time and/or the trait `embedded_hal::digital::v1::OutputPin` is not implemented.
> 
I believe this has to be a bug, as you need "some type" unless we use the `!` notype but that is unstable.

So the easiest way to do this will be to implement `OutputPin` over a newtype that does nothing.

Something like:

```rust
struct NoPin;

impl embedded_hal::digital::v2::OutputPin for NoPin {
    type Error = ();

    fn set_high(&mut self) -> Result<(), Self::Error> {
        Ok(())
    }

    fn set_low(&mut self) -> Result<(), Self::Error> {
        Ok(())
    }
}
```

followed by `builder.init(&mut delay, None::NoPin);`

This is a good case of missing ergonomics (and a bug in the docs/examples). I wonder if a `NoPin` should be provided by the embedded-hal directly though.

yanshay, 2023-11-09:
I used this solution and it worked for the example provided in this crate.
However when I tried with more complex application I had a compilation error that required me to replace `Error = ()` with `Error = Infallible` . Not sure why, but it worked :-)

Also, it should be `builder.init(&mut delay, None::<NoPin>);` 
```
struct NoPin;

impl embedded_hal::digital::v2::OutputPin for NoPin {
    type Error = Infallible;

    fn set_high(&mut self) -> Result<(), Self::Error> {
        Ok(())
    }

    fn set_low(&mut self) -> Result<(), Self::Error> {
        Ok(())
    }
}
```

rfuest, 2023-11-09:
If we decide to merge #81 it will be easier to use displays without a connected reset pin, because the reset pin it is no longer a `init` parameter. And a noop implementation of the `OutputPin` trait is already included in the create.

yanshay, 2023-11-10:
> And a noop implementation of the OutputPin trait is already included in the create

In the create or in the crate? If in the crate, where is it? I couldn't find one.

rfuest, 2023-11-10:
> > And a noop implementation of the OutputPin trait is already included in the create
> 
> In the create or in the crate? If in the crate, where is it? I couldn't find one.

Sorry, that part of my comment was a bit confusing. I meant crate (create was a typo), but only after #81 gets merged.

sampaioletti, 2024-01-11:
specific to esp, if you dont want to create a new type it is implemented for Pin driver, just pick an arbitrary pin.

```rust
None::<esp_idf_hal::gpio::PinDriver<esp_idf_hal::gpio::Gpio18, esp_idf_hal::gpio::Output>>
```

almindor, 2024-03-02:
The pin is now handled by `Builder::reset_pin`. To initialize without it just omit this call in your builder routine.

closed 2024-03-02 by almindor