almindor/mipidsi #74: Inverted Colors on st7789

kbleeke, 2023-09-30:
I upgraded from 0.6 to 0.7 and now the display colors are inverted.

This is easy enough to fix by setting ` .with_invert_colors(mipidsi::ColorInversion::Inverted)` 

Was this an intentional change?

rfuest, 2023-09-30:
> Was this an intentional change?

Yes and no. The change for the ST7789 was unintentional. I seem to have accidentally changed it as part of #52.

We have planned to default to color inversion off for all displays to be consistent across different models (see #48). But if this would have been an intentional change it would have been announced in the Changelog.

almindor, 2023-09-30:
Yes this was a bit of a mess. The issue is that displays have this set in the hw level and they differ even with same model you can get two displays with different inversion. I didn't realize this before and i thought I could tie the color inversion choice to the Model.

The final result will be a non-inverted default for all models with inversion being user choice.

kbleeke, 2023-10-01:
Thanks for you work! As I said, it was easy to adjust. Just wanted to let you know.

https://www.ebay.de/itm/124037047140 this is the display I have

closed 2024-01-30 by almindor