almindor/mipidsi #83: Ported example to ESP32 - should it be slowwww?

yanshay, 2023-11-06:
I was able to port the example 'spi-ili9486-esp32-c3' to esp32 (not c3) with ili9341 240x320 (module that include both ESP32 and display).

The example draws really slow, about 3 sec to clear the screen and another 3sec to draw the smily.
Is this how it should be or should I configure something differently. 
Only thing related to timing I changed was to changed the frequency from 100kHz to 60Mhz as I saw in other code bases for this hw config. With 100kHz it was way way slower. When I increased more than 60Mhz I didn't see any difference.

I know the display is much faster because c++ code I have us pretty fast.
But maybe the example should be slow because of how the drawing takes place?

rfuest, 2023-11-06:
Did you run the code in release mode?

almindor, 2023-11-06:
There are essentially 4 things to consider when it comes to speed:
1. Release mode [done]
2. SPI clock speed setting (if using hw SPI)
3. MCU clock speed
4. Optimizations such as blitting

1 - 3 are all in your hands, 4 is somewhere between the driver and embedded-graphics and what you draw and how.

As for your timings, when you say clearing the screen you mean via `display.clear()` ? That should preferably be "the fastest possible" in terms of your set SPI and MCU speeds given that it should really just push pixels after one window operation call.

yanshay, 2023-11-07:
I increased MCU clock speed (I thought by default it would be at max speed), and it indeed increased the speed.
And increasing SPI clock it indeed increased speed as well.
How do I know how much I can increase SPI? For MCU it's in the spec, but SPI I couldn't find, and noticed on another application that increasing speed to much draws screen distorted.
I also noticed it's possible to initialize spi with or without cs/miso, does that make a difference in performance/correctness of data transfer?

almindor, 2023-11-07:
> I increased MCU clock speed (I thought by default it would be at max speed), and it indeed increased the speed. And increasing SPI clock it indeed increased speed as well. How do I know how much I [...]

Correct, the max SPI speed (outside of the MCU itself) is per-device spec. Depends on your concrete display, it should be mentioned there.

> I also noticed it's possible to initialize spi with or without cs/miso, does that make a difference in performance/correctness of data transfer?

MISO is unused atm. (reads are planned at some point tho), CS only matters if you share your SPI bus with multiple slave devices. Going CS-less technically could increase the speed by a very small margin due to HW "delays" between CS asserts but it'll probably be in nanoseconds per second kind of thing.

rfuest, 2023-11-07:
> How do I know how much I can increase SPI? For MCU it's in the spec, but SPI I couldn't find, and noticed on another application that increasing speed to much draws screen distorted.

The official maximum SPI clock for the ILI9341 is only 10 MHz (see ILI9341 datasheet section 19.3.4), but it normally works with much higher clock rates. Other controllers, like the ST7789, support clock rates up to 62.5 MHz.

almindor, 2023-11-09:
I'm closing this issue, in case you feel there's still some problem related to `mipidsi` itself and its speed please reopen and add more details.

closed 2023-11-09 by almindor

joyel24, 2025-01-25:
@almindor I'm exeperimenting with and ESP32-S3 acutally...
It seems that display.clear() is very slow, it's taking something like half a second to clear. So I don't know what is wrong.

I beleave I set the cpu to max 160MHz and SPI to 62.5MHz.

My device is a Lilygo T-deck using an ST7789.

My actual experiment is here: [beleave](https://github.com/joyel24/t-deck-rust/blob/3912b4c68e8d379cef671fa516e5fe2be5cf4b58/src/bin/main.rs)

yanshay, 2025-01-26:
Are you building in release mode?

joyel24, 2025-01-26:
I tried with both release and debug and didn't feel any difference.

joyel24, 2025-01-26:
If I only do this 3 lines of code in a loop. What it looks like :

```rust
display.clear(Rgb565::BLACK);

let character_style = MonoTextStyle::new(&FONT_10X20, Rgb565::WHITE);
let text = "Alright";
Text::with_alignment(text, display.bounding_box().center() + Point::new(0, 15),character_style, Alignment::Center, ).draw(&mut display);
```

https://github.com/user-attachments/assets/d55dfec8-f8d0-44f2-8144-84aa69862e0f