almindor/mipidsi #87: Cannot access rest of framebuffer due to 'window_offset'

xgroleau, 2023-11-22:
We currently cannot access the total of the frame buffer due to the `window_offset`.

With this configration, when we cannot draw in the first 80 bytes since if we write at  Y address 0, the offset cause to write to the Y address 80.
```rs
        let st7789 = mipidsi::Builder::st7789(lcd_interface)
            .with_invert_colors(ColorInversion::Inverted)
            .with_display_size(240, 240)
            .with_framebuffer_size(240, 320)
            .with_orientation(mipidsi::Orientation::PortraitInverted(false))
            .with_window_offset_handler(|o| match o.orientation() {
                mipidsi::Orientation::Portrait(_) => (0, 0),
                mipidsi::Orientation::PortraitInverted(_) => (0, 80),
                mipidsi::Orientation::Landscape(_) => (0, 0),
                mipidsi::Orientation::LandscapeInverted(_) => (80, 0),
            })
            .init(
                &mut Delay,
                Some(Output::new(lcd_rst, Level::High, OutputDrive::Standard)),
            );

```
This makes it impossible to use `set_scroll_offest` with those first 80 bytes.

We could internally handle this case and trigger two draws in two writes. 

I'll try to come up with a PR

almindor, 2023-11-22:
If you need to use the extra bufferspace I think offset should then be 0 no? Use the scroll offset instead to "set the right viewport" initially if it needs to be bumped in certain orientations.

Perhaps I'm misunderstanding the setup and intent here.

xgroleau, 2023-11-22:
That would work, I wanted to avoid to let the UI know that we need an offset to the `set_scroll_offset` when doing a smooth scroll, but it's probably simpler to just set it to the right viewport and provide that viewport offset info to the UI.

I'll close the issue then

closed 2023-11-22 by xgroleau