almindor/mipidsi #88: builder.init() confuses VSCode?

shoutmarble, 2023-12-18:
I'm trying to draw a color on my Adafruit ESP32S3 TFT that has a display of 240x135

But, I can't use the display.clear(..) to set the background color because Rust doesn't know it's type?

[image]

I used [espup](https://github.com/esp-rs/espup) and [cargo-generate](https://github.com/cargo-generate/cargo-generate) the template to create a skeleton project and flashing it with [espflash](https://github.com/esp-rs/espflash).

I installed [espflash](https://github.com/esp-rs/espflash/releases/tag/v2.1.0) and [espmonitor](https://github.com/esp-rs/espmonitor).

And the remaining tools:

    cargo install espflash
    cargo install cargo-espflash
    cargo install cargo-espmonitor
    cargo install cargo-generate 
    cargo install espup 
    espup install
    cargo generate esp-rs/esp-template

Generate Skeleton Project

    PS C:\Users\XXXXX\Documents\GIT> cargo generate esp-rs/esp-template
    ⚠️   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
    🤷   Project Name: feather
    🔧   Destination: C:\Users\XXXXX\Documents\GIT\feather ...
    🔧   project-name: feather ...
    🔧   Generating template ...
    ✔ 🤷   Which MCU to target? · esp32s3
    ✔ 🤷   Configure advanced template options? · true
    ✔ 🤷   Enable allocations via the esp-alloc crate? · false
    ✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · false
    ✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · false
    ✔ 🤷   Add CI files for GitHub Action? · false
    ✔ 🤷   Setup logging using the log crate? · false
    ✔ 🤷   Enable WiFi/Bluetooth/ESP-NOW via the esp-wifi crate? · false
    🔧   Moving generated files into: `C:\Users\XXXXX\Documents\GIT\feather`...
    🔧   Initializing a fresh Git repository
    ✨   Done! New project created 

[Adafruit ESP32-S3 TFT Feather](https://learn.adafruit.com/adafruit-esp32-s3-tft-feather)

[image]

[image]

[image]

Rust Version/Toolchains

    PS C:\Users\XXXXX\Documents\GIT\feather> rustup show
    Default host: x86_64-pc-windows-msvc
    rustup home:  C:\Users\XXXXX\.rustup
    
    installed toolchains
    --------------------
    
    stable-x86_64-pc-windows-msvc (default)
    nightly-x86_64-pc-windows-msvc
    esp
    
    active toolchain
    ----------------
    
    esp (overridden by 'C:\Users\XXXXX\Documents\GIT\feather\rust-toolchain.toml')
    rustc 1.74.0-nightly (98053514f 2023-12-01) (1.74.0.1)

Rust Version:

    PS C:\Users\XXXXX\Documents\GIT\feather> cargo version
[... 242 lines cut ...]
I (51) boot: chip revision: v0.1
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (59) boot.esp32s3: SPI Mode       : DIO
I (64) boot.esp32s3: SPI Flash Size : 4MB
I (69) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 003f0000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=08564h ( 34148) map    
I (129) esp_image: segment 1: paddr=0001858c vaddr=3fc89888 size=00004h (     4) load
I (129) esp_image: segment 2: paddr=00018598 vaddr=40378000 size=01888h (  6280) load   
I (139) esp_image: segment 3: paddr=00019e28 vaddr=00000000 size=061f0h ( 25072) 
I (152) esp_image: segment 4: paddr=00020020 vaddr=42000020 size=2d194h (184724) map
I (200) boot: Loaded app from partition at offset 0x10000
I (201) boot: Disabling RNG early entropy source...
Hello world!
Loop...0
Loop...1

```

There are probably many more serious errors, this is my first go at using Rust on my micro-controller.

shoutmarble, 2023-12-18:
VS Code is now able to see type of "display" type object on Line 58
```
let mut display = builder.init(&mut delay, Some(reset)).unwrap();
```

I now have to figure out the correct PINS

[image]

```
#![no_std]
#![no_main]
#![allow(warnings)]

use display_interface_spi::{SPIInterface, SPIInterfaceNoCS};
use embedded_graphics::pixelcolor::{Rgb565, RgbColor};
use esp_backtrace as _;
use esp_println::println;

use hal::{
    clock::ClockControl,
    gpio::{GpioPin, Output, PushPull},
    peripheral,
    peripherals::Peripherals,
    peripherals::SPI2,
    prelude::*,
    spi::{self, master::Spi, FullDuplexMode, SpiMode},
    Delay,
};
use mipidsi::{
    error::InitError,
    models::{Model, ST7789},
    Builder, Display, Orientation,
};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();

    let mut clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    let io = hal::IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let sclk = io.pins.gpio36;
    let miso = io.pins.gpio37;
    let mosi = io.pins.gpio35.into_push_pull_output();
    let mut cs = io.pins.gpio7;

    let mut spi =
        hal::spi::master::Spi::new(peripherals.SPI2, 100u32.kHz(), SpiMode::Mode0, &mut clocks)
            .with_pins(Some(sclk), Some(mosi), Some(miso), Some(cs));

    // let mut i2c = io.pins.gpio21.into_push_pull_output();
    let dc = io.pins.gpio39.into_push_pull_output();
    let mut backlight = io.pins.gpio45.into_push_pull_output();
    let mut reset = io.pins.gpio40.into_push_pull_output();
    // let mut sda = io.pins.gpio42.into_push_pull_output();
    // let mut scl = io.pins.gpio41.into_push_pull_output();

    ///
    let di = SPIInterfaceNoCS::new(spi, dc);

    let builder = Builder::st7789_pico1(di);

    reset.set_high().unwrap();

    let mut display = builder.init(&mut delay, Some(reset)).unwrap();

    backlight.set_high().unwrap();
    let colors = [Rgb565::RED, Rgb565::GREEN, Rgb565::BLUE];
    display.set_pixels(240, 135, 0, 0, colors);
    // display.set_pixels(0, 0, 240, 135, colors);

    println!("Hello world!");
    let mut count = 0;
    loop {
        println!("Loop...{count}");
        count = count + 1;
        delay.delay_ms(500u32);
    }
}

```

Cargo.toml no change
```
[package]
name = "feather"
version = "0.1.0"
authors = ["Terry"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
hal = { package = "esp32s3-hal", version = "0.14.0" }
esp-backtrace = { version = "0.9.0", features = ["esp32s3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32s3"] }
mipidsi = "0.7.1"
embedded-graphics = "0.8.1"
display-interface = "0.4.1"
display-interface-spi = "0.4.1"
```

VS Code can now see's the type of Display (Line 58)
[image]

I'm most likely not using the correct pins
[image]

The schematic of the [Adafruit ESP32S3 TFT](https://cdn-learn.adafruit.com/assets/assets/000/114/104/original/adafruit_products_FS3TFT_sch.png?1660251030) and the [Pinouts on their website](https://github.com/adafruit/Adafruit-ESP32-S3-TFT-Feather-PCB/blob/main/Adafruit%20ESP32-S3%20TFT%20Feather%20Pinout.pdf) are not exactly the same.

shoutmarble, 2023-12-18:
Setting the GPIO 21 pin to high is critical (as you mentioned it)

[image]

```
#![no_std]
#![no_main]
#![allow(warnings)]

use display_interface_spi::{SPIInterface, SPIInterfaceNoCS};
use embedded_graphics::pixelcolor::{Rgb565, RgbColor};
use esp_backtrace as _;
use esp_println::println;

use hal::{
    clock::ClockControl,
    gpio::{GpioPin, Output, PushPull},
    peripheral,
    peripherals::Peripherals,
    peripherals::SPI2,
    prelude::*,
    spi::{self, master::Spi, FullDuplexMode, SpiMode},
    Delay,
};
use mipidsi::{
    error::InitError,
    models::{Model, ST7789},
    Builder, Display, Orientation,
};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();

    let mut clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    let io = hal::IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let sclk = io.pins.gpio36;
    let miso = io.pins.gpio37;
    let mosi = io.pins.gpio35.into_push_pull_output();
    let mut cs = io.pins.gpio7;
    let mut lcd_pwr = io.pins.gpio21.into_push_pull_output();

    let mut spi =
        hal::spi::master::Spi::new(peripherals.SPI2, 100u32.kHz(), SpiMode::Mode0, &mut clocks)
            .with_pins(Some(sclk), Some(mosi), Some(miso), Some(cs));

    let dc = io.pins.gpio39.into_push_pull_output();
    let mut backlight = io.pins.gpio45.into_push_pull_output();
    let mut reset = io.pins.gpio40.into_push_pull_output();

    lcd_pwr.set_high().unwrap();
    
    let di = SPIInterfaceNoCS::new(spi, dc);

    let builder = Builder::st7789_pico1(di);

    reset.set_high().unwrap();

    let mut display = builder.init(&mut delay, Some(reset)).unwrap();

    backlight.set_high().unwrap();
    let colors = [Rgb565::RED, Rgb565::GREEN, Rgb565::BLUE];
    display.set_pixels(240, 135, 0, 0, colors);
    // display.set_pixels(0, 0, 240, 135, colors);

    println!("Hello world!");
    let mut count = 0;
    loop {
        println!("Loop...{count}");
        count = count + 1;
        delay.delay_ms(500u32);
    }
}

```

[image]

shoutmarble, 2023-12-18:
I think it's working I believe that was all there was.  Thank you!
[image]

I tried also with blue to see if the color would show more with the camera
[image]

It's hard to see in this color-reduced GIF (the yellow is actually a bright red)
[image]

```
#![no_std]
#![no_main]
#![allow(warnings)]

use display_interface_spi::{SPIInterface, SPIInterfaceNoCS};
use embedded_graphics::{
    draw_target::DrawTarget,
    geometry::Point,
    pixelcolor::{Rgb565, Rgb666, RgbColor},
    primitives::Circle,
};
use esp_backtrace as _;
use esp_println::println;

use hal::{
    clock::ClockControl,
    gpio::{GpioPin, Output, PushPull},
    peripheral,
    peripherals::Peripherals,
    peripherals::SPI2,
    prelude::*,
    spi::{self, master::Spi, FullDuplexMode, SpiMode},
    Delay,
};
use mipidsi::{
    error::InitError,
    models::{Model, ST7789},
    Builder, Display, Orientation,
};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();

    let mut clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    let io = hal::IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let sclk = io.pins.gpio36;
    let miso = io.pins.gpio37;
    let mosi = io.pins.gpio35.into_push_pull_output();
    let mut cs = io.pins.gpio7;
    let mut lcd_pwr = io.pins.gpio21.into_push_pull_output();

    let mut spi =
        hal::spi::master::Spi::new(peripherals.SPI2, 100u32.kHz(), SpiMode::Mode0, &mut clocks)
            .with_pins(Some(sclk), Some(mosi), Some(miso), Some(cs));

    let dc = io.pins.gpio39.into_push_pull_output();
    let mut backlight = io.pins.gpio45.into_push_pull_output();
    let mut reset = io.pins.gpio40.into_push_pull_output();

    lcd_pwr.set_high().unwrap();

    let di = SPIInterfaceNoCS::new(spi, dc);

    let builder = Builder::st7789_pico1(di);

    reset.set_high().unwrap();

    let mut display = builder.init(&mut delay, Some(reset)).unwrap();

    backlight.set_high().unwrap();
    display.clear(Rgb565::RED).unwrap();
    backlight.set_low().unwrap();
    display.clear(Rgb666::RED.into()).unwrap();

    println!("Hello world!");
    let mut count = 0;
    loop {
        println!("Loop...{count}");
        count = count + 1;
        delay.delay_ms(500u32);
    }
}
```
[image]

shoutmarble, 2023-12-18:
Confirmation, this can be closed.

[Adafruit ESP32-S3 TFT Feather](https://learn.adafruit.com/adafruit-esp32-s3-tft-feather)

[image]

main.rs
```
#![no_std]
#![no_main]
// #![allow(warnings)]

use display_interface_spi::SPIInterfaceNoCS;
use embedded_graphics::{geometry::Point, image::Image, Drawable};
use esp_backtrace as _;
use esp_println::println;

use hal::{clock::ClockControl, peripherals::Peripherals, prelude::*, spi::SpiMode, Delay};
use mipidsi::{Builder, Orientation};
use tinybmp::Bmp;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();

    let mut clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    let io = hal::IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let sclk = io.pins.gpio36;
    let miso = io.pins.gpio37;
    let mosi = io.pins.gpio35.into_push_pull_output();
    let cs = io.pins.gpio7;
    let mut lcd_pwr = io.pins.gpio21.into_push_pull_output();

    let spi =
        hal::spi::master::Spi::new(peripherals.SPI2, 100u32.kHz(), SpiMode::Mode0, &mut clocks)
            .with_pins(Some(sclk), Some(mosi), Some(miso), Some(cs));

    let dc = io.pins.gpio39.into_push_pull_output();
    let mut backlight = io.pins.gpio45.into_push_pull_output();
    let mut reset = io.pins.gpio40.into_push_pull_output();

    lcd_pwr.set_high().unwrap();

    let di = SPIInterfaceNoCS::new(spi, dc);

    let builder = Builder::st7789_pico1(di);

    reset.set_high().unwrap();

    let mut display = builder.init(&mut delay, Some(reset)).unwrap();

    display
        .set_orientation(Orientation::Portrait(true))
        .unwrap();

    backlight.set_high().unwrap();
    let bmp_data = include_bytes!("../jesus.bmp");
    let bmp = Bmp::from_slice(bmp_data).unwrap();

    Image::new(&bmp, Point::new(0, 0))
        .draw(&mut display)
        .unwrap();

    println!("Hello world!");
    let mut count = 0;
    loop {
        println!("Loop...{count}");
        count = count + 1;
        delay.delay_ms(500u32);
    }
}
```

Cargo.toml
```
[package]
name = "feather"
version = "0.1.0"
authors = ["Terry"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
hal = { package = "esp32s3-hal", version = "0.14.0" }
esp-backtrace = { version = "0.9.0", features = ["esp32s3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32s3"] }
mipidsi = "0.7.1"
embedded-graphics = "0.8.1"
display-interface = "0.4.1"
display-interface-spi = "0.4.1"
tinybmp = "0.5.0"
profont = "0.7.0"
```

[image]

closed 2023-12-18 by shoutmarble

rfuest, 2023-12-19:
Nice to see it working. If you haven't already done it yourself, I would suggest that you increase the SPI clock rate a bit to get faster updates. The ST7789 supports up to 62.5 MHz SPI clock, but I'm not sure about the ESP32.

shoutmarble, 2023-12-19:
That is much quicker and no penalties seen yet!

[image]

[image]

```
#![no_std]
#![no_main]
// #![allow(warnings)]

use display_interface_spi::SPIInterfaceNoCS;
use embedded_graphics::{geometry::Point, image::Image, Drawable};
use esp_backtrace as _;
use esp_println::println;

use hal::{clock::ClockControl, peripherals::Peripherals, prelude::*, spi::SpiMode, Delay};
use mipidsi::Builder;
use tinybmp::Bmp;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();

    let mut clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    let io = hal::IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let sclk = io.pins.gpio36;
    let miso = io.pins.gpio37;
    let mosi = io.pins.gpio35.into_push_pull_output();
    let cs = io.pins.gpio7;
    let mut lcd_pwr = io.pins.gpio21.into_push_pull_output();

    let spi =
        hal::spi::master::Spi::new(peripherals.SPI2, 60u32.MHz(), SpiMode::Mode0, &mut clocks)
            .with_pins(Some(sclk), Some(mosi), Some(miso), Some(cs));

    let dc = io.pins.gpio39.into_push_pull_output();
    let mut backlight = io.pins.gpio45.into_push_pull_output();
    let mut reset = io.pins.gpio40.into_push_pull_output();

    lcd_pwr.set_high().unwrap();

    let di = SPIInterfaceNoCS::new(spi, dc);

    let builder = Builder::st7789_pico1(di);

    reset.set_high().unwrap();

    let mut display = builder.init(&mut delay, Some(reset)).unwrap();

    backlight.set_high().unwrap();
    let bmp_data = include_bytes!("../jesus.bmp");
    let bmp = Bmp::from_slice(bmp_data).unwrap();

    Image::new(&bmp, Point::new(0, 0))
        .draw(&mut display)
        .unwrap();

    println!("Hello world!");
    let mut count = 0;
    loop {
        println!("Loop...{count}");
        count = count + 1;
        delay.delay_ms(500u32);
    }
}
```

```
[package]
name = "feather"
version = "0.1.0"
authors = ["Terry"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
hal = { package = "esp32s3-hal", version = "0.14.0" }
esp-backtrace = { version = "0.9.0", features = ["esp32s3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32s3"] }
mipidsi = "0.7.1"
embedded-graphics = "0.8.1"
display-interface = "0.4.1"
display-interface-spi = "0.4.1"
tinybmp = "0.5.0"
profont = "0.7.0"
```