almindor/mipidsi #99: Fix examples before next release

rfuest, 2024-01-23:
We need to remember to fix the examples before the next release. They will already have been broken by the removal of the re-exports, but I would suggest to wait until we have decided which breaking PRs we want to land before the next release.

almindor, 2024-02-19:
I believe this is the last blocker for v0.8 release.

rfuest, 2024-02-19:
The migration guide also needs to be finished before the release, it is at least missing the unified defaults for color inversion and the bumped MSRV. And we must also find a solution for #76 to fix the troubleshooting link in the new version of the docs.

I'll take some time to look at #81 in the next few days. I think that the most important changes from that PR have been merged in other smaller PRs already. But it's been so long since I've looked at that code that I can't remember the details. If I find anything useful I'll open a new PR for that change and close #81.

almindor, 2024-10-21:
I believe this was done

closed 2024-10-21 by almindor