almindor/mipidsi #113: Removing the `with_` prefix from builder methods

rfuest, 2024-02-03:
Should we remove the `with_` prefix from the builder methods? The API guidelines state that the `with_` prefix should be used for constructors (https://rust-lang.github.io/api-guidelines/naming.html#casing-conforms-to-rfc-430-c-case) and builders usually don't use and prefix for the methods (https://rust-lang.github.io/api-guidelines/type-safety.html?highlight=builder#builders-enable-construction-of-complex-values-c-builder).

The naming is confusing in the latest release where [`Builder::with_model`](https://docs.rs/mipidsi/latest/mipidsi/struct.Builder.html#method.with_model) is a constructor, while other methods, like [`Builder::with_orientation`](https://docs.rs/mipidsi/latest/mipidsi/struct.Builder.html#method.with_orientation), are used to set a parameters. In the master branch `with_model` was renamed into `new`, which makes it harder to mix up the two types of methods, but I would still prefer to remove the prefix.

almindor, 2024-02-03:
Hmm so you mean it'd look like `Builder::new(<model>).orientation(Orientation::Portrait)...` ? 
I'm a bit on the fence. Something like `Builder::new(...).orientation(Orientation::Portrait)` seems a bit odd to me

rfuest, 2024-02-04:
Yes, that's what I meant. It is the recommended way of naming builder methods and should be familiar to many users, because it is used in many places like https://doc.rust-lang.org/std/process/struct.Command.html, https://docs.rs/clap/latest/clap/struct.Arg.html or https://docs.rs/embedded-graphics/latest/embedded_graphics/text/struct.TextStyleBuilder.html.

commit 8f4e01b3 (2024-02-15): Merge pull request #116 from almindor/builder-without

remove `with_` prefix from Builder, fixes #113
commit d14ab9a3 (2024-02-15): remove `with_` prefix from Builder, fixes #113
closed 2024-02-15 by almindor