almindor/mipidsi #118: TearingEffect on ST7789

nishanthkarthik, 2024-02-22:
Hi! I am a noob so please bear with me :)

Does setting `TearingEffect` make a difference when I write a bunch of pixels using `set_pixels`? I tried both `Vertical` and `HorizontalAndVertical` but I still see tearing when I alternate between black and white to fill the entire screen.

I am using ST7789 with a 135x240 display on ESP32. The tearing effect line mode number 1 (section 8.15.1 of ST7789VW datasheet) looks like what I want but I am not sure how to access this using the API.

rfuest, 2024-02-22:
`set_tearing_effect` will only enable the tearing effect output on the controller and not change how `set_pixels` works. You will need to manually synchronize the display update with the tearing effect output (see section 8.15.3 and 8.15.4 in the datasheet). Unfortunately the tearing effect output isn't available on most cheap display modules.

nishanthkarthik, 2024-02-23:
Thanks, makes sense. I'll close the issue since my display module probably doesn't support it.

closed 2024-02-23 by nishanthkarthik