almindor/mipidsi #138: `set_pixels` method seemingly not doing anything

abayomi185, 2024-06-20:
Hi, I'm using slint with mipidsi and attempting to use the `set_pixels` method to update the display with no luck.

Other methods that use draw seem to work fine. Would you happen to know if I'm doing something wrong?

I am able to see the response from TestImage and clear.

Thanks

```rust
    let di = display_interface_spi::SPIInterface::new(spi_device, dc);

    let mut display = mipidsi::Builder::new(mipidsi::models::ST7789, di)
        .reset_pin(rst)
        .invert_colors(mipidsi::options::ColorInversion::Inverted)
        // .display_size(constants::display::W as u16, constants::display::H as u16)
        .init(&mut delay)
        .unwrap();

    // Clear the display
    display.clear(Rgb565::BLACK).unwrap();
    display
        .set_orientation(mipidsi::options::Orientation::new())
        .unwrap();

    mipidsi::TestImage::new().draw(&mut display).unwrap(); // Works

    display.clear(Rgb565::RED).unwrap(); // This works too
    display
        .set_pixels(
            0,
            0,
            240,
            320,
            &mut [slint::platform::software_renderer::Rgb565Pixel(0);
                constants::display::W as usize]
                .iter()
                .map(|x| embedded_graphics::pixelcolor::raw::RawU16::new(x.0).into()),
        )
        .unwrap();
```

rfuest, 2024-06-20:
The `ex` and `ey` coordinates are part of an inclusive range. Try this:

```rust
    display
        .set_pixels(
            0,
            0,
            239,
            319,
            &mut [slint::platform::software_renderer::Rgb565Pixel(0);
                constants::display::W as usize]
                .iter()
                .map(|x| embedded_graphics::pixelcolor::raw::RawU16::new(x.0).into()),
        )
        .unwrap();
```

abayomi185, 2024-06-21:
Sadly no difference with this change

abayomi185, 2024-06-21:
`set_pixel` looks to work fine.

rfuest, 2024-06-21:
If `set_pixel` works `set_pixels` should also work, because they use very similar code:
https://github.com/almindor/mipidsi/blob/7ff5271e20b5d835e8ecc27537a49b5f55ec3e99/mipidsi/src/lib.rs#L198-L204
https://github.com/almindor/mipidsi/blob/7ff5271e20b5d835e8ecc27537a49b5f55ec3e99/mipidsi/src/lib.rs#L225-L240

Your code does only draw a single black line on top of the display, which might make it hard to see during debugging. Does this work `display.set_pixels(0, 0, 239, 319, core::iter::repeat(Rgb565::MAGENTA).take(240 * 320))?;`?

abayomi185, 2024-06-21:
Thanks so much! This works. It seems I have an issue elsewhere.

closed 2024-06-21 by abayomi185