almindor/mipidsi #140: Is the support available for esp32s3 gc9a01 board

MohaBeacon, 2024-07-02:
tried building it with slint example - didn't work as expected.

https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.28

rfuest, 2024-07-02:
You'll need to provide us with additional information, like:
- What didn't work exactly? Did the code not compile, did the display show nothing or did the display show a corrupted image?
- Did you check everything that is mentioned in https://github.com/almindor/mipidsi/blob/master/docs/TROUBLESHOOTING.md?
- Which version of mipidsi are you using?
- What code did you use to initialize the display?
- Does the problem also occur if you don't use slint, e.g. by drawing the test image that is included in the `mipidsi` crate:
  ```rust
  TestImage::new().draw(&mut display).unwrap();
  ```

almindor, 2024-10-21:
Closing as inactive, please feel free to reopen.

closed 2024-10-21 by almindor