almindor/mipidsi #141: st7789 blank screen

GustavoKatel, 2024-09-21:
the usual: I'm very new to esp-rs, esp32 and such, so bear with me

- my screen does not have a CS pin
- I can see the backlight turning on, but nothing else
- Tried playing around with different baudrate
- Tried MODE_0 and MODE_3
- I tried following the other related issues

none of the above worked :( 

any tips on how to troubleshoot this?

Here's my code:

```rust
    let mut rst = PinDriver::output(peripherals.pins.gpio0)?;
    rst.set_high()?;

    let dc = PinDriver::output(peripherals.pins.gpio1)?;
    let mut delay = Ets;

    let mut backlight = PinDriver::output(peripherals.pins.gpio10)?;
    backlight.set_high()?;

    let scl = peripherals.pins.gpio2;
    let spi = peripherals.spi2;
    let sda = peripherals.pins.gpio4;

    let spi = SpiDriver::new(
        spi,
        scl,
        sda,
        None::<AnyIOPin>,
        &DriverConfig {
            // dma: Dma::Auto(240 * 240 * 2 + 8),
            ..Default::default()
        },
    )?;

    let cs: Option<AnyIOPin> = None;

    let spi = SpiDeviceDriver::new(
        spi,
        cs,
        &SPIConfig {
            baudrate: 26.MHz().into(),
            data_mode: MODE_0,
            ..Default::default()
        },
    )?;

    let di = SPIInterface::new(spi, dc);
    let mut display = Builder::new(models::ST7789, di)
        .reset_pin(rst)
        .display_size(240, 240)
        .init(&mut delay)
        .map_err(|err| anyhow!("Error initializing display: {:?}", err))?;

    // Clear the display to red
    info!("Clearing display...");
    display
        .clear(Rgb565::RED)
        .map_err(|err| anyhow!("Error clearing display: {:?}", err))?;
    info!("Display cleared!");

    info!("Displaying image...");
    // Clear the display initially
    display.clear(Rgb565::GREEN).unwrap();

    // Draw a rectangle on screen
    let style = embedded_graphics::primitives::PrimitiveStyleBuilder::new()
        .fill_color(Rgb565::GREEN)
        .build();

    embedded_graphics::primitives::Rectangle::new(Point::zero(), display.bounding_box().size)
        .into_styled(style)
        .draw(&mut display)
        .unwrap();
    info!("Displaying image... DONE");
```

rfuest, 2024-09-21:
Those ST7789 displays without a CS pin can be really annoying and shouldn't exist, but you can usually get them to work by using `MODE_3`.

I don't see anything wrong with your code at first glance. If you like you can upload the entire project somewhere and I can try to reproduce the issue.

GustavoKatel, 2024-09-21:
@rfuest that would be really helpful!

here's a cleanup code: https://github.com/GustavoKatel/esp-rs-st7789-test

btw forgot to mention I'm using an esp32c3 super mini

tested this repro and same result. unfortunately I do not have other board to test :(

rfuest, 2024-09-21:
I've just tried your code and it did work without any changes:
[image]
For the picture I've changed the code a bit to draw mipidsi's `TestImage` and enabled color inversion in the builder call to get the correct colors. But even without those changes the display worked and showed a solid color.

I guess that either your display module was DOA or something with your wiring isn't correct.

GustavoKatel, 2024-09-22:
@rfuest thank you so much for taking the time to test the code. I managed to get another display and it is indeed working as expected. 

> Those ST7789 displays without a CS pin can be really annoying and shouldn't exist, but you can usually get them to work by using MODE_3.

I believe this seems to be the case for this issue :/

closed 2024-09-22 by GustavoKatel