almindor/mipidsi #142: How to speed things up (ST7789V2 + ESP32-C6-DevKitC-1)

akx, 2024-10-18:
Heya,

I'm wondering how to speed things up for an application that will likely need full-screen updates most of the time.

I have an [ESP32-C6-DevKitC-1](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html) and a [WaveShare 280x240 1.69" LCD module](https://www.waveshare.com/wiki/1.69inch_LCD_Module), using `esp-idf-svc` (at present anyway).

My current experiment code (please excuse the mess, it's an experiment so far) is

```rust
use std::collections::VecDeque;
use std::time::Instant;
use esp_idf_svc::hal::gpio::PinDriver;
use esp_idf_svc::hal::prelude::*;
use crate::led::WS2812RMT;

use embedded_graphics::{
    prelude::*,
    primitives::Rectangle,
};
use embedded_hal::spi::MODE_3;
use esp_idf_svc::hal::delay::Ets;
use esp_idf_svc::hal::spi::config::{BitOrder, Config, Duplex};
use esp_idf_svc::hal::spi::{SpiDeviceDriver, SpiDriverConfig};
use display_interface_spi::SPIInterface;
use embedded_graphics::pixelcolor::{Rgb565, Rgb888};
use mipidsi::models::ST7789;
use mipidsi::options::{ColorInversion, Orientation, RefreshOrder, Rotation};
use rand::{Rng, SeedableRng};
use rand::rngs::SmallRng;
use crate::ticktock::Ticktock;

mod led;
mod ticktock;

fn main() -> anyhow::Result<()> {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();
    log::info!("Hello, world!");
    let peripherals = Peripherals::take()?;
    log::info!("Initializing LED");
    let mut led = WS2812RMT::new(peripherals.pins.gpio8, peripherals.rmt.channel0)?;
    led.set_pixel(50, 50, 0)?;
    let mut rng = SmallRng::from_entropy();
    log::info!("Initializing pins");
    let spi = peripherals.spi2;
    let mut backlight = PinDriver::output(peripherals.pins.gpio23)?;
    let rst = PinDriver::output(peripherals.pins.gpio22)?;
    let dc = PinDriver::output(peripherals.pins.gpio21)?; // DC (data/command)
    let sclk = peripherals.pins.gpio19; // CLK
    let sda = peripherals.pins.gpio18; // DIN
    let sdi = peripherals.pins.gpio15; // Tied low
    let cs = peripherals.pins.gpio20; // CS
    log::info!("Initializing SPI device");
    let mut delay = Ets;
    let config = Config::new()
        .bit_order(BitOrder::MsbFirst)
        .data_mode(MODE_3)
        .baudrate(80.MHz().into());

    let device = SpiDeviceDriver::new_single(
        spi,
        sclk,
        sda,
        Some(sdi),
        Some(cs),
        &SpiDriverConfig::new(),
        &config,
    )?;
    log::info!("Initializing display_interface_spi");
    let di = SPIInterface::new(device, dc);
    log::info!("Initializing display");
    let mut display = mipidsi::Builder::new(ST7789, di)
        .reset_pin(rst)
        .display_size(240, 320)
        .orientation(Orientation::new().rotate(Rotation::Deg90))
        .invert_colors(ColorInversion::Inverted)
        .init(&mut delay).unwrap();
    log::info!("Setting BL");
    backlight.set_high()?;

    let mut sw = Ticktock::new();
    let mut buf: Vec<Rgb565> = Vec::with_capacity(320*240);
    let alloc_time = sw.tick();
    for i in 0..240 {
        for j in 0..320u16 {
            let val = i ^ (j as u8);
            let px = Rgb888::new(val, val, val);
            buf.push(Rgb565::from(px));
        }
    }
    let initial_fill_time = sw.tick();
    log::info!("Allocated buffer in {alloc_time}us, filled buffer in {initial_fill_time}us",
        alloc_time = alloc_time,
        initial_fill_time = initial_fill_time,
    );

    let mut frame_times: VecDeque<u32> = VecDeque::with_capacity(20);
    let fullscreen = Rectangle::new(Point::new(0,0), Size::new(320,240));
    let mut last_fps_print_time = Instant::now();
    for _i in 1..10000 {
        let mut sw = Ticktock::new();
        for _ in 1..100 {
            let inoff = rng.gen_range(0..buf.len());
            let outoff = rng.gen_range(0..buf.len());
            buf.swap(inoff, outoff);
        }
        let render_time = sw.tick();
        let clon = buf.clone();
        let clone_time = sw.tick();
        display.fill_contiguous(&fullscreen, clon).unwrap();
        let blit_time = sw.tick();
        let frame_time = sw.total();
        frame_times.push_back((frame_time / 1000) as u32);
        if frame_times.len() > 20 {
            frame_times.pop_front();
        }
        if last_fps_print_time.elapsed().as_millis() > 500 {
            led.set_pixel(0, 50, 0)?;
            let fps = 1000.0 / (frame_times.iter().sum::<u32>() as f32 / (frame_times.len() as f32));
            log::info!(
                "FPS: {fps:.2} / render={render_time}us clone={clone_time}us blit={blit_time}us frame={frame_time}us",
                fps = fps,
                render_time = render_time,
                clone_time = clone_time,
                blit_time = blit_time,
                frame_time = frame_time,
            );
            last_fps_print_time = Instant::now();
            led.set_pixel(0, 0, 50)?;
        }
    }
    unreachable!();
}
```
and the interesting (performance) output for a `--release` build is:

```
I (742) telsu: Allocated buffer in 35us, filled buffer in 25514us
I (1292) telsu: FPS: 9.26 / render=120us clone=649us blit=107889us frame=108662us
I (1832) telsu: FPS: 9.26 / render=122us clone=649us blit=107897us frame=108672us
```

IOW, 99.2% of the frame time is spent in `fill_contiguous`. Setting the SPI baudrate to something lower than 80 MHz (which, AIUI, is already pushing it especially given my display is behind 10-centimeter DuPont wires 🤠) doesn't change things a lot; blit time becomes about 123547us).

Is there something obvious I'm doing wrong for an application like this, where I basically just have a buffer of Rgb565 to push to the screen?

almindor, 2024-10-18:
Hmm I don't see anything particularly wrong in the code, but AFAICS  you should be able to push roughly 60fps with 80Mhz SPI. Of course in practice that's going to be a bit lower (and I only did my napkin math real quick I might've made a mistake here).

For one full screen of filler you're doing 320x240x2x8 bits so `1_228_800` bits to be pushed over at 80Mhz should fit roughly 65 of those a second.

One thing I do not see though in your code is setting the MCU's processing speed/clocks. I don't know how ESP32 inits but I remember that on my hifive1 I had to set my clock to full. In my case it's 320Mhz, in yours it's 160Mhz from what I see. The default was something quite lower IIRC. It might be you're just running on lower clocks on your MCU.

rfuest, 2024-10-18:
The `display-interface` crate and `embedded-hal` 1.0 add quite a bit of overhead, which slows down the display update rate. To work with the new `SpiDevice` trait `display-interface-spi` buffers the data in 64 byte chunks before it passes it to the HAL. And to make things worse the CS pin is toggled for each chunks. Combined with the unnecessary dynamic dispatch in `display-interface` the performance suffers noticeable for fast displays.

almindor, 2024-10-18:
> The `display-interface` crate and `embedded-hal` 1.0 add quite a bit of overhead, which slows down the display update rate. To work with the new `SpiDevice` trait `display-interface-spi` buffers the [...]

Hmm that's quite unfortunate. Do you think we should abandon `display-interface` use? I'm also curious why 64 byte chunks are necessary? I don't remember that being any kind of requirement when we discussed the e-h 1.0 SPI traits.

akx, 2024-10-19:
Thank you for the response and thoughts.

> One thing I do not see though in your code is setting the MCU's processing speed/clocks. I don't know how ESP32 inits but I remember that on my hifive1 I had to set my clock to full. In my case it's [...]

The full log (which of course I should've posted in the first place, but [here](https://gist.github.com/akx/a15f779103845d034b1aa47c0110fa9a)) says

```
I (220) cpu_start: Pro cpu start user code
I (221) cpu_start: cpu freq: 160000000 Hz
```

so unless something in the ESP IDF drops that down after the app starts, I think I should be running at 160MHz.

akx, 2024-10-19:
> I'm also curious why 64 byte chunks are necessary?

That seems to be [this bit of code](https://github.com/therealprof/display-interface/blob/8fca041b0288740678f16c1d05cce21bd3867ee5/spi/src/lib.rs#L81-L103).

Looks like the intent is to read a 64-byte chunk from the pixel iterator, endian-flip it and send it off.

almindor, 2024-10-19:
> > I'm also curious why 64 byte chunks are necessary?
> 
> That seems to be [this bit of code](https://github.com/therealprof/display-interface/blob/8fca041b0288740678f16c1d05cce21bd3867ee5/spi/src/lib.rs#L81-L103).
> 
> Looks like the intent is to read a 64-byte chunk from the pixel iterator, endian-flip it and send it off.
> 
> **EDIT:** Changing that buffer size in a local copy of `display-interface-spi`:
> 
> buffer size	result
> 64	`FPS: 9.26 / render=120us clone=649us blit=107343us frame=108116us`
> 128	`FPS: 9.90 / render=123us clone=649us blit=100722us frame=101498us`
> 256	`FPS: 10.31 / render=119us clone=649us blit=96933us frame=97705us`
> 512	`FPS: 10.42 / render=122us clone=650us blit=95296us frame=96072us`
> 1024	`FPS: 10.53 / render=123us clone=650us blit=94953us frame=95730us`
> 3840	Guru Meditation Error: Core 0 panic'ed (Stack protection fault). 😄

Perfect, I was hoping to see how this goes. It seems the flipflopping of buffer isn't the main slowdown here. Making the buffer bigger improves it but not in a linear fashion at all, considering the 1024 case should've been 16 time better in theory and is < 10% improvement.

Could you try changing fill_congiguous into a fill_solid, or possibly calling set_pixels directly? I just want to make sure it's not something funky.

rfuest, 2024-10-20:
I did some testing and I think the ESP HAL has quite a bit of overhead for the kind of transfers we use. I'm currently testing on a ESP32-C3 with the no_std HAL. With some tweaks I was able to get the time to clear the framebuffer down to ~60ms, which still isn't great.

Using a logic analyzer did reveal some interesting patterns:
[image]
For this capture I've used a `display-interface-spi` buffer size of 256 bytes and a SPI clock of 40 MHz (I did set 60 MHz in my program, but I measured 40 MHz). The HAL splits up every 256 byte buffer into four 64 byte blocks, because that is the SPI FIFO size (https://docs.esp-rs.org/esp-hal/esp-hal/0.21.1/esp32c3/esp_hal/spi/master/struct.Spi.html#method.write_bytes). While the 64 byte sub blocks are transferred at full SPI clock speed without any gaps between bytes, the gaps between the 64 byte blocks cause a significant overhead. This alone causes the write performance to drop by ~33%.

I'm not familiar enough with the ESP32-C3 internals to know if this dead time could be reduced. The "best" solution would probably be a framebuffer in RAM, which is then transferred via DMA to the display, but that does require a lot of RAM and there is currently no support for DMA in this create and the embedded-graphics ecosystem in general.

akx, 2024-10-20:
> Could you try changing fill_congiguous into a fill_solid, or possibly calling set_pixels directly? I just want to make sure it's not something funky.

I changed the draw bit to
```rust
display.fill_solid(&fullscreen, if _i % 2 == 0 { Rgb565::BLACK } else { Rgb565::WHITE }).unwrap();
```

```
I (1347) telsu: FPS: 10.10 / render=125us clone=4us blit=99427us frame=99560us
I (1947) telsu: FPS: 10.10 / render=121us clone=4us blit=99424us frame=99552us
I (2547) telsu: FPS: 10.10 / render=125us clone=4us blit=99425us frame=99558us
I (3147) telsu: FPS: 10.10 / render=124us clone=4us blit=99426us frame=99558us
```

which is still kind of unfortunate (and funky, I think?)...

akx, 2024-10-21:
@rfuest Nice work! Am I understanding correctly that since the `Framebuffer` is marked `BigEndian`, the price for converting from native (little-endian) pixel formats to big-endian is paid when drawing to the buffer instead of when transferring, and since `draw_fast` just does a "trust me bro, these bytes are what we want to push out to the bus", that works out? 😄

rfuest, 2024-10-21:
Yes, that's correct. Some API changes will be necessary to make this safer to use without the chance of accidentally using the wrong endianness.

In my code the CPU is often just waiting for the DMA transfer to finish and converting a little-endian framebuffer into a big-endian format during that time wouldn't negatively impact the time it takes to update the display. But the `display-interface` crate doesn't provide a way to specify the endianness of a `&[u8]` buffer. This [hack](https://gist.github.com/rfuest/487f0ee3adb67acd1da2138eaf751ac9) works and is as fast as before, but it is on a whole other level of "trust me bro".

almindor, 2024-10-21:
So from what I understand there are a few issues here:
- `display-interface` forcing 64 bit buffer size
- `display-interface` forcing endianness runtime conversion
- `display-interface` using dynamic dispatch

Optionally I think another issue is that:
- `embedded-hal` not providing a native iterator SPI `write/transfer/read` methods

I think if `embedded-hal` actually had iterator support, and if we figured out a better abstraction for `display-interface` we should be able to get a fairly quick write using an iterator with 0 additional buffering?

rfuest, 2024-10-21:
> display-interface forcing 64 bit buffer size

`display-interface` doesn't force any buffer size, but the implementation for SPI in `display-interface-spi` does use a 64 word buffer to convert the iterator based `DataFormat`s into a buffer of bytes that can be passed to the `SpiDevice` impl.

> display-interface forcing endianness runtime conversion

No it doesn't force runtime conversion, but It only support different endianness with `u16` buffers with the `DataFormat::{U16BE, U16LE}` variants. In my opinion it is inconvenient that you need to cast a more common `&[u8]` buffer to `&[u16]` first, which to my knowledge requires an external crate like `byte-slice-cast` (which the `display-interface` impls use internally).

> * `display-interface` using dynamic dispatch

This has been an issue for a long time and there is even an open draft PR from 2021 with a possible solution in the `display-interface` repo, but nobody seems to have cared about that: https://github.com/therealprof/display-interface/pull/19

I'm not sure if we (or someone) should fix `display-interface`, which seems to be more or less unmaintained, or if it is better to create a new alternative. In any case this will need some careful design to get this right.

> * `embedded-hal` not providing a native iterator SPI `write/transfer/read` methods
>
> I think if `embedded-hal` actually had iterator support, and if we figured out a better abstraction for [...]

This might help for some drawing operations, but I'm not sure it is needed. Randomly drawing individual pixels over a relatively slow connection like SPI always comes with a lot of overhead. For images we shouldn't use iterators at all.

I'm planning to add a something like `draw_image(&mut self, image: &ImageRaw<...>, position: Point)` to e-g's `DrawTarget` trait, which will allow `DrawTarget` implementations to use a more efficient code path in that case. And because `ImageRaw` is just a thin wrapper around a byte slice, that specifies the image dimensions, color format, and byte order, this could also be used for other applications like writing an entire RAM framebuffer to a display. I'm still not sure about the details, but ideally this addition to e-g and the changes to `display-interface` would make it possible to use DMA transfers for displaying images in more cases. And this might also be a stepping stone towards async support.

fu5ha, 2024-10-21:
If you see my luluu repo, I went through the process of getting within a few percent of the theoretical bandwidth limits of the raspi 2040 spi device. There is some interesting stuff in the main firmware but in particular see the patches in the vendored crates, including the rp-hal implementation and mipidsi.

It's not clear to me how much these "fixes" can or should actually be upstreamed but maybe it's useful for reference https://github.com/fu5ha/luluu/tree/main/software

almindor, 2024-10-21:
@rfuest Lmk what you think about https://github.com/almindor/mipidsi/pull/143

@fu5ha Thanks I'll have a look!

almindor, 2024-10-26:
@akx could you please try the `Display::set_pixels_from_buffer()` to see how much of a difference it makes?

akx, 2024-11-03:
@almindor Sorry, I was away from this project for a while 😅 

Not a great improvement (current [mess of a code here](https://gist.github.com/akx/f7039f4fe726eb2ebec07c87cb21a485)):

```
$ cargo espflash flash --monitor -p /dev/cu.usbmodem1101 --release
[2024-11-03T16:59:22Z INFO ] 🚀 A new version of cargo-espflash is available: v3.2.0
[2024-11-03T16:59:22Z INFO ] Serial port: '/dev/cu.usbmodem1101'
[2024-11-03T16:59:22Z INFO ] Connecting...
[2024-11-03T16:59:22Z INFO ] Using flash stub
   Compiling telsu v0.1.0 (/Users/akx/build/telsu)
    Finished `release` profile [optimized] target(s) in 1.39s
Chip type:         esp32c6 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       f0:f5:bd:01:64:8c
Bootloader:        /Users/akx/build/telsu/target/riscv32imac-esp-espidf/release/build/esp-idf-sys-86f39b5293325c54/out/build/bootloader/bootloader.bin
Partition table:   part.csv
App/part. size:    462,960/4,194,304 bytes, 11.04%
[2024-11-03T16:59:24Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-11-03T16:59:24Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:03] [========================================]     250/250     0x10000                                                                                     [2024-11-03T16:59:28Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4080053c
0x4080053c - rmt_driver_isr_default
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xe2c
load:0x4086e610,len:0x2e24
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2.2 2nd stage bootloader
I (23) boot: compile time Nov  3 2024 18:50:31
I (24) boot: chip revision: v0.1
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (30) boot.esp32c6: SPI Mode       : DIO
I (35) boot.esp32c6: SPI Flash Size : 8MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00400000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=4c05ch (311388) map
I (156) esp_image: segment 1: paddr=0005c084 vaddr=40800000 size=03f94h ( 16276) load
I (160) esp_image: segment 2: paddr=00060020 vaddr=42050020 size=175d8h ( 95704) map
I (181) esp_image: segment 3: paddr=00077600 vaddr=40803f94 size=08070h ( 32880) load
I (190) esp_image: segment 4: paddr=0007f678 vaddr=4080c010 size=019d4h (  6612) load
I (195) boot: Loaded app from partition at offset 0x10000
I (196) boot: Disabling RNG early entropy source...
I (210) cpu_start: Unicore app
W (218) clk: esp_perip_clk_init() has not been implemented yet
I (225) cpu_start: Pro cpu start user code
I (225) cpu_start: cpu freq: 160000000 Hz
I (226) cpu_start: Application information:
I (228) cpu_start: Project name:     libespidf
I (233) cpu_start: App version:      a2f4705
I (238) cpu_start: Compile time:     Nov  3 2024 18:50:25
I (244) cpu_start: ELF file SHA256:  000000000...
I (250) cpu_start: ESP-IDF:          v5.2.2
I (254) cpu_start: Min chip rev:     v0.0
[... 16 lines cut ...]
I (367) main_task: Started on CPU0
I (367) main_task: Calling app_main()
I (367) telsu: Hello, world!
I (377) telsu: Initializing LED
I (377) telsu: Initializing pins
I (377) gpio: GPIO[23]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (387) gpio: GPIO[22]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (397) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (407) telsu: Initializing SPI device
I (417) telsu: Initializing display_interface_spi
I (417) telsu: Initializing display
I (727) telsu: Setting BL
I (767) telsu: Allocated buffer in 48us, filled buffer in 41370us
I (1367) telsu: Tick: 6, FPS: 13.89 / render=123us blit=72075us frame=72202us
I (1877) telsu: Tick: 13, FPS: 13.89 / render=121us blit=72078us frame=72203us
I (2387) telsu: Tick: 20, FPS: 13.89 / render=121us blit=72081us frame=72206us
I (2887) telsu: Tick: 27, FPS: 13.89 / render=122us blit=72077us frame=72203us
I (3397) telsu: Tick: 34, FPS: 13.89 / render=122us blit=72077us frame=72203us
I (3907) telsu: Tick: 41, FPS: 13.89 / render=121us blit=72082us frame=72207us
I (4407) telsu: Tick: 48, FPS: 13.89 / render=121us blit=72078us frame=72203us
I (4917) telsu: Tick: 55, FPS: 13.89 / render=124us blit=72075us frame=72203us
I (5417) telsu: Tick: 62, FPS: 13.89 / render=125us blit=72074us frame=72203us
I (5927) telsu: Tick: 69, FPS: 13.89 / render=124us blit=72075us frame=72203us
I (6437) telsu: Tick: 76, FPS: 13.89 / render=123us blit=72080us frame=72207us
```

almindor, 2024-11-03:
> Not a great improvement

That's unfortunate. I'd expect a much better improvement. I got myself esp32-C6, once I have some time I'll try to experiment myself and see where the holdups are.

almindor, 2024-11-09:
Ok so for me, using the `esp32-C6` with SPI set to `80Mhz` and a `320x240` `ST7789` display with CS:

* I can get to `~30ms` fullscreen draw time when using the new `Display::set_pixels_from_buffer` so roughly 33FPS, still slow but a bit better.
* Using `fill_solid` drops down to `~70ms` so worse than twice slowdown!
* Using `fill_contiguous` I wasn't able to create a full framebuffer with `Rgb565`, but using a `.copied().cycled()` iterator it's even slightly worse than `fill_solid` at about `~72ms`

So at least on the C6 this clearly shows that the buffering/abstraction we currently depend on is a big problem, but not the only issue.

I'm going to try and see if I can get closer to the C speed (pun intended) by removing the 64byte buffer problem in the `display-interface` directly next. We should be able, when using `set_pixels_from_buffer` and no "mid-re-buffering" able to get the full ~60 FPS speeds expected.

almindor, 2024-11-09:
I did some more testing and if I deconstruct the SPI back after sending all the init stuff and prepping a data window, I'm able to send the raw buffer directly using the SPI, without and `display-interface` involvement, but it still clocks at exactly the same speed as the `set_pixels_from_buffer`.

This means that the SPI implementation of the `esp32-hal` (I'm not using IDF) is probably sub-optimal somehow.

Full code [here](https://gist.github.com/almindor/2a3c7243d4aa32e886895e145c7dc6d1)
NOTE: you need `prep_pixels_from_buffer` added to `mipidsi` for this to work, the code for that is also in the GIST at the bottom. It's just a hack to get this going for testing, the basic idea is to see if raw SPI direct u8 buffer transfer is the culprit here, which it seems to be.

almindor, 2024-11-09:
@rfuest @akx Found it!  Note I used only `esp-hal` not the IDF/std abstraction.

The main issues are:
1. CPU default of esp-hal is 80Mhz, so make sure to set it to 160Mhz manually
2. Spi by default will not use DMA buffers, you need to set it up explicitly
3. Using the new `Display::set_pixels_from_buffer` is cruicial due to what we discussed originally

The code for the "fixed up" version is [here](https://gist.github.com/almindor/d693ae9fa6edc71463afa161da01f775)

The result now looks like this:
```
INFO - solid drawing time 44062 us
INFO - from_buffer drawing time 16103 us
INFO - direct drawing time 16050 us
```

Which means ~62.5 FPS.

I'll keep this issue opened until we release `mipidsi` with the new `set_pixels_from_buffer` method. The `display-interface` situation still needs thinking over as well of course as it does cause a major slowdown.

markrossington, 2025-09-07:
Hello all. Great investigation here, appreciate the hard work.

 I notice that `set_pixels_from_buffer` doesn't exist any more. What is the fastest way to send pixels in the current API? Thank you.