almindor/mipidsi #150: STM32H573I-DK display with ST7789 not working (blank)

chrenderle, 2024-11-22:
Hello together,

I have the STM32H573I-DK which has a 240x240 ST7789 Display with 8016 Parallel transfer. I want to use the display with mipidsi and I could verify that the display was working because the demo firmware was using it. Sadly the demo firmware source code is not freely available so I couldn't compare it.

I tried all the steps in the troubleshooting section but it still doesn't work. I also checked that all the pins are correct multiple times. I don't know what I can do else to get it to run. 

You can find my code here: https://github.com/chrenderle/stm32h573i-dk
The datasheet for the display here: http://frdlcd.cn/Upload/1628131252.pdf
The user manual for the discovery kit here: https://www.st.com/resource/en/user_manual/um3143-discovery-kit-with-stm32h573ii-mcu-stmicroelectronics.pdf
The schematic for the discovery kit here: https://www.st.com/resource/en/schematic_pack/mb1677-h573i-c02-schematic.pdf

Parsifa1, 2025-04-06:
i have the same issue: the screen only blank with white background (due to the `bl.set_high()`), but i've tested it with some c demo and there is nothing wrong with screen itself:(
this is my code:
```rust
#![no_std]
#![no_main]

use defmt::info;
use embassy_executor::Spawner;
use embassy_stm32::gpio::{Level, Output, Pin, Speed};
use embassy_time::{Delay, Duration, Timer};
use embedded_graphics::{pixelcolor::Rgb565, prelude::*};
use mipidsi::{
    interface::{Generic16BitBus, ParallelInterface},
    models::ST7789,
    Builder,
};
use {defmt_rtt as _, panic_probe as _};

macro_rules! lcd_output {
    ($($pin:expr),*) => {
        ($( Output::new($pin.degrade(), Level::High, Speed::VeryHigh),)*)
    };
}

// Embassy STM32 配置
#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    info!("Hello, world!");
    let p = embassy_stm32::init(Default::default());
    let pins = lcd_output!(
        p.PB0, p.PB1, p.PB2, p.PB3, p.PB4, p.PB5, p.PB6, p.PB7, p.PB8, p.PB9, p.PB10, p.PB11,
        p.PB12, p.PB13, p.PB14, p.PB15
    );
    let bus = Generic16BitBus::new(pins);
    let wr = Output::new(p.PC7.degrade(), Level::High, Speed::VeryHigh); // LCD_WR -> PC7
    let dc = Output::new(p.PC8.degrade(), Level::High, Speed::VeryHigh); // LCD_RS -> PC8

    let mut _cs = Output::new(p.PC9.degrade(), Level::High, Speed::VeryHigh);
    let mut rd = Output::new(p.PC6.degrade(), Level::High, Speed::VeryHigh);
    let mut _bl = Output::new(p.PC10.degrade(), Level::Low, Speed::VeryHigh); // LCD_BL -> PC5
    let di: ParallelInterface<_, _, _> = ParallelInterface::new(bus, dc, wr);
    rd.set_high();

    let mut delay = Delay;

    let mut display = Builder::new(ST7789, di)
        .display_size(240, 320)
        .color_order(mipidsi::options::ColorOrder::Rgb)
        .init(&mut delay)
        .unwrap();
    _bl.set_high();

    display.clear(Rgb565::GREEN).unwrap();
    loop {
        Timer::after(Duration::from_millis(500)).await;
    }
}
```

rfuest, 2025-04-07:
I unfortunately don't have access to a 16 bit parallel display to try to reproduce the issue myself.

Are you using some kind of development board or custom hardware, and can you provide more information like schematics or the C code you used?

> the RST of the lcd was connect with the RST of the board it self, so i cant set hard RST. maybe this is the problem?

It shouldn't, but it might, because I think that most users are using hardware reset.

Parsifa1, 2025-04-07:
[tftlcd.zip](https://github.com/user-attachments/files/19626821/tftlcd.zip)

here is the example c code.
it is GBK decoded, so maybe the comment are 
Garbled characters in UTF-8

Thank you very much！

rfuest, 2025-04-07:
I did a quick browse through the C code and noticed a comment about a quirk of the STM32F103s, where some pins aren't usable without disabling the JTAG interface. Here is the relevant section of the HAL docs about this: https://docs.rs/stm32f1xx-hal/latest/stm32f1xx_hal/gpio/index.html#accessing-pa15-pb3-and-pb14