almindor/mipidsi #158: Also supports ILI9488

NotABadger, 2025-01-12:
No problem here, Just wanted to mention that I've tested this on an ILI9488 with the mipidsi::models::ILI9486Rgb666 instance using the Generic8BitBus. and BRG color order.
Thanks you for contributing to open source!

rfuest, 2025-01-17:
Thanks for letting us know.

@almindor: Should we add a ILI9488 type to make this an officially supported model? It's a fairly common IC and at the moment it wouldn't be much of a maintenance burden. Having a separate type would allow us to add a `Rgb888` mode in the future, which isn't available in the ILI9486.

NotABadger, 2025-01-18:
Today I had a morning to hobby around, I've also tried the SPI interface, which works.
As a (relatively) young embedded engineer not experienced in display busses, I wanted to "drag race"  the different interfaces, so I wrote a test program, executed it on the MIPI type DBI 8-bit (as mentioned above) which took up to 7.1 seconds.
After that I tried to do the same via the SPI interface, but I didn't known which clock frequency to use so I then read the datasheets of the ILI9488, and ILI9486.  Which also don't mention a frequency or speed. The ESP32-S3 can run up to 80 MHz according to the Datasheet. After some experimenting I managed to run it up to 60 MHz on a breadboard with the ILI9488, and complete the same test program in 1.42 sec (and I accidentally drag raced RGB565 vs RGB666 on SPI, because I didn't read the SPI part of the datasheet beforehand).

I was wondering, if I could toy with the clock frequency of the generic bus and then I read the ili9486.rs file, and found that 120_000 us (or 120 ms) is used between write cycles.  May I know where that value came from? I'm not criticizing or anything, just curious about these type of interfaces and trying to learn. Feel free to just comment with links/articles/datasheets. 

Below is my drag race code in case you are interested, this is running on an ESP32-S3 using the esp-idf framework.

```
fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    esp_idf_svc::sys::link_patches();
    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Display test");

    //Setting up:
    // Data/Command (D/C) output pin, Reset Pin.
    let periph = Peripherals::take().unwrap();
    let lcd_dc = PinDriver::output(periph.pins.gpio19).unwrap();
    let lcd_reset = PinDriver::output(periph.pins.gpio40).unwrap();
    let lcd_chip_select = periph.pins.gpio18;
    let mut delay = Delay::default();

    //Setting up spi bus
    let spi_controller = periph.spi2;
    let spi_driver = SpiDriver::new(spi_controller,
                               periph.pins.gpio15,
                               periph.pins.gpio17, 
                               Some(periph.pins.gpio16), 
                               &SpiDriverConfig::new()).expect("spi_driver init failed");
    let spi_device_config = config::Config::new().baudrate(60.MHz().into());
    let spi_device_driver = SpiDeviceDriver::new(&spi_driver, Some(lcd_chip_select), &spi_device_config).unwrap();
    
    //mipidsi lib time!
    let mut buffer = [0_u8; 512]; //Buffer for spi
    let lib_spi_abstraction = SpiInterface::new(spi_device_driver, lcd_dc, &mut buffer);
    // Create the ILI9486 display driver from the display interface with the RGB666 color space
    let mut display = Builder::new(ILI9486Rgb666, lib_spi_abstraction)
    .reset_pin(lcd_reset)
    .orientation(Orientation::default().rotate(mipidsi::options::Rotation::Deg90).flip_vertical())
    .display_size(320, 480)
    .color_order(mipidsi::options::ColorOrder::Bgr)
    .init(&mut delay).unwrap();

    loop {  
        let total_start_time = std::time::Instant::now();  
        let mut start_time = std::time::Instant::now();
        if let Err(error) = display.clear(Rgb666::BLUE) {
            log::error!("Can not clear the screen, SPI error: {:?}", error);
        }
        let mut end_time = std::time::Instant::now();
        let mut duration = end_time - start_time;
        log::info!("Blue, took {} ms", duration.as_millis());
        //delay is for humans
        //Delay::delay_ms(&Delay::new_default(), 2000);
        Delay::delay_ms(&Delay::new_default(), 1);

        start_time = std::time::Instant::now();
        _ = display.clear(Rgb666::RED);
        end_time = std::time::Instant::now();
        duration = end_time- start_time;
        log::info!("Red, took {} ms", duration.as_millis());
        //Delay::delay_ms(&Delay::new_default(), 2000);
        Delay::delay_ms(&Delay::new_default(), 1);

        //should be left top
        start_time = std::time::Instant::now();
        _ = display.fill_solid(&Rectangle::new(Point { x: 0, y: 0 }, Size { width: 100, height: 50 }), Rgb666::GREEN);
[... 41 lines cut ...]
        //Does not adhere to rotation!
        start_time = std::time::Instant::now();
        _ = display.set_vertical_scroll_region(0, 0);
        _ = display.set_vertical_scroll_offset(100);
        end_time = std::time::Instant::now();
        duration = end_time- start_time;
        log::info!("scroll up, took {} ms", duration.as_millis());
        // Delay::delay_ms(&Delay::new_default(), 2000);
        Delay::delay_ms(&Delay::new_default(), 1);
        
        start_time = std::time::Instant::now();
        _ = display.set_vertical_scroll_region(0, 0);
        _ = display.set_vertical_scroll_offset(0);
        end_time = std::time::Instant::now();
        let total_end_time = std::time::Instant::now();
        duration = end_time- start_time;
        log::info!("scroll down, took {} ms", duration.as_millis());
        // Delay::delay_ms(&Delay::new_default(), 2000);
        Delay::delay_ms(&Delay::new_default(), 1);
        
        let total_duration = total_end_time - total_start_time;
        log::info!("total test, took {} ms", total_duration.as_millis());
    }
}
```

almindor, 2025-01-18:
> I was wondering, if I could toy with the clock frequency of the generic bus and then I read the ili9486.rs file, and found that 120_000 us (or 120 ms) is used between write cycles. May I know where [...]

Do you mean [this delay](https://github.com/almindor/mipidsi/blob/master/src/models/ili9486.rs#L35)?

That's an "initialization pause" that most displays need in order to actually register a reset initiated just before [here](https://github.com/almindor/mipidsi/blob/master/src/builder.rs#L163-L173). The reset can be "soft" or "hard" depending on if a reset pin is provided or if there's a cmd used to reset the display. In either case it takes a bit of time for the display to come back.

The 120ms time was taken from SOMEWHERE but I'll be first to honestly say I can't remeber where and if it's not a copy pasta when it comes to differences between different models. E.g. it wouldn't surprise me if some model can do it faster than others.

It is only done on re-inits though which shouldn't be in the hot path?

NotABadger, 2025-01-18:
You're completely right. Sorry, I was reading the code wrong. 
I have to admit, your solution is pretty neat. Although I will have to start learning the Macro language in Rust to fully understand it. 
One last question then (I promise I'll stop after), I can't spot any other delay/wait/frequency in your code. Is this solution I/O speed bottle necked? Is the pin switching determining the transfer speed?

rfuest, 2025-01-18:
> The 120ms time was taken from SOMEWHERE but I'll be first to honestly say I can't remeber where and if it's not a copy pasta when it comes to differences between different models. E.g. it wouldn't surprise me if some model can do it faster than others.

We previously had some issues with unreliable inits on the ILI934x, because the delays were too short. The init code for that model contains some references to the relevant datasheet sections, which should be similar to the ILI9488: https://github.com/almindor/mipidsi/blob/3f92dc83857f726d9e3c14f4f801fbd90727a045/src/models/ili934x.rs#L13. But as @almindor already said, this is probably not your issue.

> After that I tried to do the same via the SPI interface, but I didn't known which clock frequency to use so I then read the datasheets of the ILI9488, and ILI9486. Which also don't mention a frequency or speed.

The datasheets mention this indirectly. The table in section 17.4.3 of the ILI9488 datasheet contains the minimum serial clock cycle, which is 50 ns. This means the official maximum SPI clock frequency is only 20 MHz. In practice much higher clock frequencies also work, but I wouldn't use this in a professional application.

rfuest, 2025-01-18:
> I have to admit, your solution is pretty neat. Although I will have to start learning the Macro language in Rust to fully understand it.
> One last question then (I promise I'll stop after), I can't [...]

In case of the generic parallel GPIO interface each pin is set individually via the HAL, which isn't very efficient. The CPU overhead will be higher than just writing one value into the SPI peripheral, which will take care of all the rest.

The runtimes you mentioned in your previous post seem very high. Just to be sure: Are you running in release mode?

NotABadger, 2025-01-18:
@rfuest Thank you for looking into it as well. I was typing my last response, and when I clicked "comment" I saw your response as well. 
Thank you for pointing out the timing part of the datasheet. This is my first exploration / deep dive of busses & communication. These days everything has a library, but I am/was curious how the magic of the implementation was created. 
@almindor & @rfuest Thank you for answering my questions, and guiding be down this rabbit hole!

rfuest, 2025-01-18:
> [@almindor](https://github.com/almindor) & [@rfuest](https://github.com/rfuest) Thank you for answering my questions, and guiding be down this rabbit hole!

No problem, feel free to ask if you have any more questions.

NotABadger, 2025-01-18:
> The runtimes you mentioned in your previous post seem very high. Just to be sure: Are you running in release mode?

Yes, I am building with the release flag. 
With the test program above (now set to 20 MHz), I use cargo run --release > test_spi_results.txt, and generate the following output: 
 -- Flashing and booting -- 
` [0;32mI (394) main_task: Started on CPU0[0m
[0;32mI (404) main_task: Calling app_main()[0m
[0;32mI (404) screen_test: Display test[0m
[0;32mI (404) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (414) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1004) screen_test: Blue, took 346 ms[0m
[0;32mI (1354) screen_test: Red, took 346 ms[0m
[0;32mI (1374) screen_test: Left top, took 11 ms[0m
[0;32mI (1394) screen_test: Right top, took 11 ms[0m
[0;32mI (1414) screen_test: Left Bottom, took 11 ms[0m
[0;32mI (1434) screen_test: Right Bottom, took 11 ms[0m
[0;32mI (2484) screen_test: Drawing test image, took 1044 ms[0m
[0;32mI (2494) screen_test: scroll up, took 0 ms[0m
[0;32mI (2504) screen_test: scroll down, took 0 ms[0m
[0;32mI (2514) screen_test: total test, took 1840 ms[0m ` 
-- loop repeats --

rfuest, 2025-01-18:
I've just tried your code on a ESP32-C3 with a ST7789 display running via SPI at 20 MHz. Here are my results:
```
INFO - Blue, took 77 ms
INFO - Red, took 77 ms
INFO - Left top, took 5 ms
INFO - Right top, took 5 ms
INFO - Left Bottom, took 5 ms
INFO - Right Bottom, took 5 ms
INFO - Drawing test image, took 236 ms
```
My display is only 320x240, which explains some of the difference. But even the corner tests, which have a fixed size of 100x50 pixels run twice as fast in my test. I left out the total runtime, because I didn't run the scroll tests which made the total invalid.

I'm using the bare metal HAL and not the IDF variant. Perhaps that could influence the performance and might be worth investigating further.

NotABadger, 2025-01-19:
I did like your hypothesis so much, that my brain decided to wake me up at 06:00 so I could work on verifying this.
however, this send me down into a new rabbit hole with a rustc coredump as a result. 

To start, I generated a new no_std project as described in the [ESP book](https://docs.esp-rs.org/book/writing-your-own-application/generate-project/index.html). I got an error that my rustc compiler was out of date, so I updated it with 'rustup update' and 'espup update', which now uses rustc 1.84.0.0 .

This made the 'hello world'  program run fine. 
Then I set out on a journey to copy paste my old program, include/use the no_std hal.
I used the example for the SPI ESP32-C3  from this repository to implement my code.
However, I got a core dump from rustc when compiling on line 65 where the SpiInterface is used.
Any idea what I did wrong?

```Rust
#![no_std]
#![no_main]

use defmt::info;
//esp_hall
use esp_hal::clock::CpuClock;
use esp_hal::delay::Delay;
use esp_hal::gpio::Output;
use esp_hal::gpio::Level;
use esp_hal::main;
use esp_hal::spi::master::{Config, Spi};
use esp_hal::spi::Mode;
use esp_hal::time::RateExtU32;
use esp_hal::time::Instant;
use esp_hal::time::now;
use esp_hal::time::Duration;
use {defmt_rtt as _, esp_backtrace as _};
//embedded_hal_bus
use embedded_hal_bus::spi::AtomicDevice;
use embedded_hal_bus::util::AtomicCell;
//mipidsi uses
use mipidsi::interface::SpiInterface;
use mipidsi::models::ILI9486Rgb666;
use mipidsi::options::Orientation;
use mipidsi::TestImage;
use mipidsi::Builder;
//embedded_graphics uses
use embedded_graphics::primitives::Rectangle;
use embedded_graphics::{prelude::*, pixelcolor::Rgb666};

extern crate alloc;
extern crate mipidsi;
extern crate embedded_graphics;

#[main]
fn main() -> ! {
    // generator version: 0.2.2

    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let periph = esp_hal::init(config);

    esp_alloc::heap_allocator!(72 * 1024);
    let delay = Delay::new();

    let lcd_dc = periph.GPIO14;
    let lcd_reset = periph.GPIO15;
    let lcd_cs = Output::new(periph.GPIO10, Level::High);
    let mut spi = Spi::new(
        periph.SPI2,
        Config::default()
            .with_frequency(20.MHz())
            .with_mode(Mode::_0),
    )
    .unwrap()
    .with_sck(periph.GPIO12)
    .with_miso(periph.GPIO11) // order matters
    .with_mosi(periph.GPIO13); // order matters
    let spi_bus_atomic_cell = AtomicCell::new(spi);
[... 43 lines cut ...]
  19:     0x7fdf5181bc7d - <rustc_data_structures[7490c3560e13ac19]::sync::parallel::ParallelGuard>::run::<(), rustc_data_structures[7490c3560e13ac19]::sync::parallel::par_for_each_in<&rustc_span[6216dc77597e7c6d]::def_id::LocalDefId, &[rustc_span[6216dc77597e7c6d]::def_id::LocalDefId], <rustc_middle[8ac56485c6056ae4]::hir::map::Map>::par_body_owners<rustc_hir_analysis[f2f24cae42e238bb]::check_crate::{closure#4}>::{closure#0}>::{closure#0}::{closure#1}::{closure#0}>
  20:     0x7fdf5161d40e - rustc_hir_analysis[f2f24cae42e238bb]::check_crate
  21:     0x7fdf4d78fe29 - rustc_interface[fa0ba49808ba01a1]::passes::analysis
  22:     0x7fdf52acbd7e - rustc_query_impl[139339c607ef3f83]::plumbing::__rust_begin_short_backtrace::<rustc_query_impl[139339c607ef3f83]::query_impl::analysis::dynamic_query::{closure#2}::{closure#0}, rustc_middle[8ac56485c6056ae4]::query::erase::Erased<[u8; 1usize]>>
  23:     0x7fdf52e4b5e8 - <rustc_query_impl[139339c607ef3f83]::query_impl::analysis::dynamic_query::{closure#2} as core[19c24f2bf64da68c]::ops::function::FnOnce<(rustc_middle[8ac56485c6056ae4]::ty::context::TyCtxt, ())>>::call_once
  24:     0x7fdf52b2577f - rustc_query_system[58145ffb095b1890]::query::plumbing::try_execute_query::<rustc_query_impl[139339c607ef3f83]::DynamicConfig<rustc_query_system[58145ffb095b1890]::query::caches::SingleCache<rustc_middle[8ac56485c6056ae4]::query::erase::Erased<[u8; 1usize]>>, false, false, false>, rustc_query_impl[139339c607ef3f83]::plumbing::QueryCtxt, true>
  25:     0x7fdf52cb50ee - rustc_query_impl[139339c607ef3f83]::query_impl::analysis::get_query_incr::__rust_end_short_backtrace
  26:     0x7fdf4d468bfe - <rustc_middle[8ac56485c6056ae4]::ty::context::GlobalCtxt>::enter::<rustc_driver_impl[11120e4604616cb8]::run_compiler::{closure#0}::{closure#1}::{closure#6}, core[19c24f2bf64da68c]::result::Result<(), rustc_span[6216dc77597e7c6d]::ErrorGuaranteed>>
  27:     0x7fdf4d4b1d65 - <rustc_interface[fa0ba49808ba01a1]::interface::Compiler>::enter::<rustc_driver_impl[11120e4604616cb8]::run_compiler::{closure#0}::{closure#1}, core[19c24f2bf64da68c]::result::Result<core[19c24f2bf64da68c]::option::Option<rustc_interface[fa0ba49808ba01a1]::queries::Linker>, rustc_span[6216dc77597e7c6d]::ErrorGuaranteed>>
  28:     0x7fdf4d533ff2 - <scoped_tls[4cdd8087fbbd96d6]::ScopedKey<rustc_span[6216dc77597e7c6d]::SessionGlobals>>::set::<rustc_interface[fa0ba49808ba01a1]::util::run_in_thread_with_globals<rustc_interface[fa0ba49808ba01a1]::util::run_in_thread_pool_with_globals<rustc_interface[fa0ba49808ba01a1]::interface::run_compiler<core[19c24f2bf64da68c]::result::Result<(), rustc_span[6216dc77597e7c6d]::ErrorGuaranteed>, rustc_driver_impl[11120e4604616cb8]::run_compiler::{closure#0}>::{closure#1}, core[19c24f2bf64da68c]::result::Result<(), rustc_span[6216dc77597e7c6d]::ErrorGuaranteed>>::{closure#0}, core[19c24f2bf64da68c]::result::Result<(), rustc_span[6216dc77597e7c6d]::ErrorGuaranteed>>::{closure#0}::{closure#0}::{closure#0}, core[19c24f2bf64da68c]::result::Result<(), r [... cut ...] ce<()>>::call_once::{shim:vtable#0}
  32:     0x7fdf5465f5bb - std::sys::pal::unix::thread::Thread::new::thread_start::hba5155a029b79976
  33:     0x7fdf49e7d148 - start_thread
  34:     0x7fdf49f010cc - __GI___clone3
  35:                0x0 - <unknown>

rustc version: 1.84.0-nightly (b0bd11bcc 2025-01-10) (1.84.0.0)
platform: x86_64-unknown-linux-gnu

query stack during panic:
#0 [typeck] type-checking `__xtensa_lx_rt_main`
#1 [analysis] running analysis passes on this crate
end of query stack

rfuest, 2025-01-19:
I was able to get the code to compile by creating `Outputs` for `lcd_cs` and `lcd_reset` instead of using `.into()`:
```rust
    let lcd_dc = Output::new(periph.GPIO14, Level::Low);
    let lcd_reset = Output::new(periph.GPIO15, Level::High);
    let lcd_cs = Output::new(periph.GPIO10, Level::High);
```

> I did like your hypothesis so much, that my brain decided to wake me up at 06:00 so I could work on verifying this.
> however, this send me down into a new rabbit hole with a rustc panic as a result.

And I wasn't quite awake anymore when I wrote reported my findings yesterday. I forgot to mention that my display was running in RGB565 mode, which would account for some of the performance difference. The performance I'm getting is also not far off the theoretical minimum transfer time at 20 MHz SPI clock.

NotABadger, 2025-01-20:
@rfuest , you might want to redo your calculations again. With your fix implemented I got the logging: "INFO  total test, took 1086 ms" going no_std vs std shaved 754ms off of the test program. That's neat! I mean 754/18,4 ms means that we shaved/cut off about 41%. That's really significant! (I reran this test 3 times before I believed it)

I did know that the std implementation took some performance off of the table, but 41% is really something and makes you reconsider. Now I have to redesign my hand held digital dice roller with no_std in mind. Thank you (again) for being open to this sort of experimentation, and supporting me. I think this was really a good learning experience. 

Although I am surprised that the ESP fork of the rustc compiler actually panicked instead of giving an error. Well, that shows it's written by humans after all.
Just know that your work is enjoyed by people like me that love to figure things out and experiment with this sort of thing.

NotABadger, 2025-01-20:
One Last addition. (I'll leave it alone after this one, I swear)
 I tried rendering some text on the screen:

no_STD:
INFO  Black, took 208 ms
└─ screen_bare_metal_spi_test::__xtensa_lx_rt_main @ src/bin/main.rs:91  
INFO  Text, took 21 ms
└─ screen_bare_metal_spi_test::__xtensa_lx_rt_main @ src/bin/main.rs:108 

STD:
I (9985) screen_test: Black, took 346 ms
I (12135) screen_test: Text, took 150 ms

It's insane! 

No STD Code: 
```Rust
#![no_std]
#![no_main]

use defmt::info;
//esp_hall
use esp_hal::clock::CpuClock;
use esp_hal::delay::Delay;
use esp_hal::gpio::Output;
use esp_hal::gpio::Level;
use esp_hal::main;
use esp_hal::spi::master::{Config, Spi};
use esp_hal::spi::Mode;
use esp_hal::time::RateExtU32;
use esp_hal::time::Instant;
use esp_hal::time::now;
use esp_hal::time::Duration;
use {defmt_rtt as _, esp_backtrace as _};
//embedded_hal_bus
use embedded_hal_bus::spi::AtomicDevice;
use embedded_hal_bus::util::AtomicCell;
//mipidsi uses
use mipidsi::interface::SpiInterface;
use mipidsi::models::ILI9486Rgb666;
use mipidsi::options::Orientation;
use mipidsi::TestImage;
use mipidsi::Builder;
//embedded_graphics uses
use embedded_graphics::primitives::Rectangle;
use embedded_graphics::{prelude::*, pixelcolor::Rgb666};
// Provides the required color type
use embedded_graphics::mono_font;
use u8g2_fonts::types::*;
use u8g2_fonts::FontRenderer;
use u8g2_fonts::fonts;

extern crate alloc;
extern crate mipidsi;
extern crate embedded_graphics;

#[main]
fn main() -> ! {
    // generator version: 0.2.2

    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let periph = esp_hal::init(config);

    esp_alloc::heap_allocator!(72 * 1024);
    let mut delay = Delay::new();

    let lcd_dc = Output::new(periph.GPIO14, Level::Low);
    let lcd_reset = Output::new(periph.GPIO15, Level::High);
    let lcd_cs = Output::new(periph.GPIO10, Level::High);
    let spi = Spi::new(
[... 121 lines cut ...]
        _ = display.clear(Rgb666::BLACK);
        let mut end_time = std::time::Instant::now();
        let mut duration = end_time - start_time;
        log::info!("Black, took {} ms", duration.as_millis());
        Delay::delay_ms(&Delay::new_default(), 2000);
        //TODO try only blacking out text instead of whole screen for performance
        start_time = std::time::Instant::now();
        // Draw the text .
        font.render_aligned(
            text,
            display.bounding_box().center() + Point::new(0, 16),
            VerticalPosition::Baseline,
            HorizontalAlignment::Center,
            FontColor::Transparent(Rgb666::CYAN),
            &mut display,
        ).unwrap();

        end_time = std::time::Instant::now();
        duration = end_time - start_time;
        log::info!("Text, took {} ms", duration.as_millis());
        Delay::delay_ms(&Delay::new_default(), 2000);
    }
}

```

closed 2025-01-22 by NotABadger