almindor/mipidsi #161: Make troubleshooting guide easier to discover

rfuest, 2025-01-14:
I think that we could make the troubleshooting guide easier to discover. In the docs the link is hidden after the examples in a very short section without any indication what kind of issues the troubleshooting guides addresses. My suggestion would be to move it up on the page and add some guidance when a user might want to check the guide.

For the link in the README i would really like to move the README back into the root of the repo, because at the moment there isn't much relevant information on the first page a user sees when they visit the repo. My suggestion would be to move the `mipidsi` crate into the root of the repo and just have `mipidsi-async` as a workspace member.

almindor, 2025-01-14:
>  My suggestion would be to move the `mipidsi` crate into the root of the repo and just have `mipidsi-async` as a workspace member.

How would that work exactly? You can only have one `Cargo.toml` in the root.

I'm happy to unify the README though, I think it should be common to both crates.

rfuest, 2025-01-14:
> How would that work exactly? You can only have one `Cargo.toml` in the root.

You can define a crate and a workspace in the same `Cargo.toml`. Take a look at e-g's `Cargo.toml` as an example, which is used for the `embedded-graphics` crate and also defines the workspace that contains the `embedded-graphics-core` crate: https://github.com/embedded-graphics/embedded-graphics/blob/38401271e86cd9097692aa0633e20e294aebec87/Cargo.toml#L20-L21.

almindor, 2025-01-14:
> You can define a crate and a workspace in the same `Cargo.toml`. Take a look at e-g's `Cargo.toml` as an example, which is used for the `embedded-graphics` crate and also defines the workspace that [...]

Ah that's great. I somehow missed that for all these years :D I'll make a PR when the time permits

almindor, 2025-01-16:
@rfuest do you think with the repackage that this issue is fixed now that README.md points to it directly or do you want to make it somehow more apparent?

rfuest, 2025-01-17:
First of all we need to fix the links to the troubleshooting document, because they were broken by moving the markdown file. I still think that we should add some kind of overview of what kind of issues are addressed in the troubleshooting document to the readme and the link in `lib.rs`. Something like "Check the troubleshoting guide if your display stays blank or the displayed colors are incorrect." would be enough IMO. In `lib.rs` I would also suggest to move the troubleshooting section above the examples.

In addition to that we could also create a GH issue template with a link to the troubleshooting guide.

almindor, 2025-01-18:
This should now be solved, feel free to re-open if you feel something more is needed. I added the issue template and relinked and moved the troubleshooting guide links + added more info around them

closed 2025-01-18 by almindor