almindor/mipidsi #164: Interface trait is too limited

ost-ing, 2025-01-17:
Hello, 

The `Interface` trait doesn't really allow much flexibility with custom strategies, or experimentation, for example its hard to know the pixel position that would be written (without extracting this from the send_command function). 

In other display implementations, there is an additional layer for custom caching and optimization strategies, for example the ssd1306 https://github.com/rust-embedded-community/ssd1306 has a `BufferedGraphics Mode`.

I'm interested in implementing some kind of Dirty Rectangle method for selective clearing of the screen, because at the moment clearing the entire screen on an stm32f4 at 42MHz, greatly reduces the FPS and introduces flicker.

rfuest, 2025-01-17:
The `Interface` trait is the wrong layer of abstraction for what you are trying to implement. You can uses something like a embedded-graphics [`Framebuffer`](https://docs.rs/embedded-graphics/latest/embedded_graphics/framebuffer/struct.Framebuffer.html) to prepare a larger buffer, which you can then send as one update to the display.

omelia-iliffe, 2025-06-25:
I am experimenting with this. Is there an easy way to draw the `FrameBuffer` to `mipidsi::Display`?

rfuest, 2025-06-25:
You can convert the e-g `Framebuffer` to an image to draw it like this:
```rust
use embedded_graphics::prelude::*;
use embedded_graphics::image::Image;
use embedded_graphics::framebuffer::Framebuffer;

let display = mipidsi::Builder::new(...).init(...)?;

let mut fb = Framebuffer::<...>::new();
mipidsi::TestImage::new().draw(&mut fb)?;

Image::new(fb.as_image(), Point::zero()).draw(&mut display)?;
```

omelia-iliffe, 2025-06-25:
Fantastic! Thank you so much!