almindor/mipidsi #167: Add physical connection type to Interface trait

rfuest, 2025-01-18:
Some display controllers support different color depth with different connection types. The ILI9488, for example, supports different color depth for SPI and parallel connections.

We've never been able to detect if the combination between physical connection and chosen color depth is valid. But now that we no longer depend on `display-interface` we could add additional meta information to the `Interface` trait to detect these kinds of configuration errors and alert the user about them.

almindor, 2025-01-20:
How do you propose to model this? A const parameter with the `Model` type-bound to a "typeflag" of implementors of the Interface trait?

rfuest, 2025-01-20:
I hadn't really thought about the details of the implementation when I opened the issue. But I think adding a const to the `Interface` trait would be a good idea like:
```rust
trait Interface {
   const KIND: InterfaceKind;
   ...
}

#[non_exhaustive]
enum InterfaceKind {
   Spi,
   Parallel8080,
}
```

The check for valid configuration would then be performed by the `Model`s `init` method. This is more flexible than trying to find a type level solution for this problem. The `init` method would need to be changed to return an `InitError` and a `InvalidConfiguration` variant added to that enum.

rfuest, 2025-02-14:
This has been implemented in #169.

closed 2025-02-14 by rfuest