almindor/mipidsi #171: [Help] SPI with DMA transfert with embassy.

mgrenonville, 2025-01-24:
Hi ! Thanks a lot for this awesome project. 

First, I have a little (and old) background in MCU, started programming them with ASM, then C/C++ with arduino. I received as a secret santa gift an esp32-c6, I though it will be a good playground for embedded rust. So, I consider myself as a beginner.

I've already read discussion about https://github.com/almindor/mipidsi/pull/149, https://github.com/almindor/mipidsi/issues/142, and https://github.com/almindor/mipidsi/issues/82, to implement correctly SPI with DMA.

I'm also using embassy, since I'm confortable with await/async (My daily language is scala). 

After doing a lot of trial/error, I ended up with mipidsi 0.9.0 and esp-hal-embassy 0.6.0 (that pulls esp-hal 0.23.1).

Currently, I've setup things like this : 
```
macro_rules! singleton {
    ($val:expr, $T:ty) => {{
        static STATIC_CELL: ::static_cell::StaticCell<$T> = ::static_cell::StaticCell::new();
        STATIC_CELL.init($val)
    }};
}
.... 
pub fn init() {
    let mut config = esp_hal::Config::default();
    config.cpu_clock = CpuClock::_160MHz;
    let peripherals = esp_hal::init(config);
 
   let dc = Output::new(peripherals.GPIO15, Level::Low);
    let sck = peripherals.GPIO18;
    let miso = peripherals.GPIO22;
    let mosi = peripherals.GPIO19;
    let cs = peripherals.GPIO4;

    // Define the reset pin as digital outputs and make it high
    let mut rst = Output::new(peripherals.GPIO6, Level::Low);
    rst.set_high();

    let (rx_buffer, rx_descriptors, tx_buffer, tx_descriptors) = dma_buffers!(3200, 3200);
    let dma_rx_buf = DmaRxBuf::new(rx_descriptors, rx_buffer).unwrap();
    let dma_tx_buf = DmaTxBuf::new(tx_descriptors, tx_buffer).unwrap();

    // Define the SPI pins and create the SPI interface
    let spi = Spi::new(
        peripherals.SPI2,
        Config::default()
            .with_frequency(60u32.MHz())
            .with_mode(Mode::_0),
    )
    .unwrap()
    .with_sck(sck)
    .with_mosi(mosi)
    .with_miso(miso)
    .with_dma(peripherals.DMA_CH0)
    .with_buffers(dma_rx_buf, dma_tx_buf);

    let cs_output = Output::new(cs, Level::High);

    let spi_device = ExclusiveDevice::new_no_delay(spi, cs_output).unwrap();

    // Define the display interface with no chip select
    let buffer = singleton!([0_u8; 240], [u8; 240]);
    let di = SpiInterface::new(spi_device, dc, buffer);
    // Define the display from the display interface and initialize it
    let mut delay = Delay::new();

    let mut display = Builder::new(ST7789, di)
        .reset_pin(rst)
        .color_order(mipidsi::options::ColorOrder::Rgb)
        .invert_colors(ColorInversion::Inverted)
        // .orientation(Orientation::new().rotate(Rotation::Deg180))
        .init(&mut delay)
        .unwrap();
...

      let start = time::now();
      display.fill_solid(&display.bounding_box(), Rgb565::BLACK).unwrap();
      let total = time::now() - start;
      log::info!("clear time {}", total);
}
```

However, my MCU background and discussion about new Spi trait makes me feel I'm missing something about `buffer` given to SpiInterface and `tx_buffer`. 
Also, even if i understand basically why it's working (with traits implementations in esp-hal), I feel like my code is copying something to a buffer that is then copied to DMA, then DMA copies to LCD. 
After some test, in release mode, clearing the entire screen is about 35ms, and I've read that with a esp32c6 at 160Mhz and spi at 80Mhz, i can expect ~15ms. Since my screen bugs at 80Mhz, I'm using 60Mhz, but It should not nearly double the time.

Also, the [gist](https://gist.github.com/almindor/d693ae9fa6edc71463afa161da01f775) you provide in one of the mentionned issue is not valid anymore in v0.9.0 (`set_pixels_from_buffer` is not defined anymore)

Speaking of this example, IIRC my MCU basics, why do we need an intermediary buffer ? Can't we use `tx_buffer` directly, to save an extra iteration over pixels ?

Thanks a lot for your replies. I'm willing to make an example with embassy to help future people facing the same issue !

georgik, 2025-01-24:
I'm also interested in the solution for DMA.
Current version of https://github.com/georgik/esp-display-interface-spi-dma/ is working only with older esp-hal 0.22 and mipidsi 0.8. I wonder if we can get the DMA support directly to mipidsi crate.

I built also some Board Support Packages macros which allows targeting most common boards, but with esp-hal 0.23.1 and mipidsi 0.9, it requires an update.

Some working examples for esp-hal 0.22 and mipidsi 0.8 are here:
- https://github.com/georgik/esp32-spooky-maze-game
- https://github.com/georgik/esp32-conways-game-of-life-rs

I would appreciate a hint how to approach this DMA topic.

mgrenonville, 2025-01-25:
Hey, 
I've already spotted your repositories ! After reading messages on matrix.org, I've spotted [this lib](https://github.com/birdistheword96/ssd1322_rs/blob/main/examples/src/bin/frames.rs). To release the full power of dma, I think that we have to make it `async` (to allow other async task execution while waiting DMA-transfer end interruption). 
With this trick, it's possible to update antoher buffer while the first is sent by DMA.
So, without an async mipidsi version, DMA will not release it's true potential

However, my remark on the transfer speed stays, I can't understand why am i twice slow than expected with blocking DMA.

almindor, 2025-01-25:
I'll look at what I get with my C6, I've been doing some work with it lately.

When it comes to async, it's hard. I did some experimenting and found out a few things wrt. going async:
1. We can do a full code duplication async version crate, but it won't work with embedded-graphics, because embedded-graphics are not async traits. We can't call async draw implementations from within `DrawTarget` implementation.
2. We can use `#[maybe_async]` to keep most code singular, but the same issue is hit with embedded-graphics
3. We can reimplement some parts to be async via a full framebuffer only, which would work with embedded-graphics since drawing would be sync, with flushing being the only async operation. I suspect this is the best and quite frankly only real answer. Note that this requires the user to flush manually after drawing, but there are caveats on how this can be done (see bellow)

The bufffered hybrid approach is hard to implement in practice however, because:
1. The DC pin needs to be set and if we keep that sync it blocks the execution loop.
2. All delay calls would also block the execution loop since we'd be keeping them sync.
3. The RST pin has the same issue and would block the execution loop.
4. To use DMA buffers correctly, the hypotetical "AsyncBufferedInterface" would have to use a single buffer for everything, which can be tricky in places.

There's a possible "workaround" hack if we keep `Model::init` sync to keep them singular, or possibly abstract them to use `#[maybe_async]` with delay and pin switch abstractions (with `#[maybe_async]`).

This would mean that no delay would be called from sync calls originating in `embedded-graphics`. We would only sync-block minimally during DC pin switching.

@rfuest what are your thoughts?

almindor, 2025-01-26:
I just tested with my esp32-C6 and I get ~18ms for full screen clear on a `ST7789`. I use this code

```rust
#![no_std]
#![no_main]

// use display_interface_spi::SPIInterface as SpiInterface;
use embedded_graphics::{
    draw_target::DrawTarget,
    pixelcolor::Rgb565,
    prelude::{Point, RgbColor, Size},
    primitives::Rectangle,
};
use embedded_hal_bus::spi::ExclusiveDevice;
use esp_backtrace as _;
use esp_hal::{
    clock::Clocks,
    delay::Delay,
    dma::{Dma, DmaPriority, DmaRxBuf, DmaTxBuf},
    dma_buffers,
    gpio::*,
    prelude::*,
    spi::*,
    time,
};
use master::Spi;
use mipidsi::{interface::SpiInterface, models::ST7789, options::*, Builder};

#[entry]
fn main() -> ! {
    #[allow(unused)]
    let mut config = esp_hal::Config::default();
    config.cpu_clock = CpuClock::Clock160MHz;
    let peripherals = esp_hal::init(config);
    let mut delay = Delay::new();

    esp_println::logger::init_logger_from_env();

    log::info!("running at {}", Clocks::get().cpu_clock);

    let dma = Dma::new(peripherals.DMA);
    let dma_channel = dma.channel0;
    let dma_buffs = dma_buffers!(32000);
    let rx_buff = DmaRxBuf::new(dma_buffs.1, dma_buffs.0).unwrap();
    let tx_buff = DmaTxBuf::new(dma_buffs.3, dma_buffs.2).unwrap();

    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);
    let mosi = io.pins.gpio23;
    let miso = io.pins.gpio13;
    let sclk = io.pins.gpio22;
    let cs_display = Output::new(io.pins.gpio21, Level::High);
    let dc = Output::new(io.pins.gpio20, Level::Low);
    let rst_display = Output::new(io.pins.gpio19, Level::Low);

    let spi = Spi::new(peripherals.SPI2, 80.MHz(), SpiMode::Mode0)
        .with_pins(sclk, mosi, miso, NoPin)
        .with_dma(dma_channel.configure(false, DmaPriority::Priority0))
        .with_buffers(rx_buff, tx_buff);
    log::info!("Spi created");

    let spi_device_display = ExclusiveDevice::new(spi, cs_display, delay).unwrap();
    log::info!("SpiDevices created");

    let mut display_buf = [0u8; 512];
    let di = SpiInterface::new(spi_device_display, dc, &mut display_buf);
    log::info!("DI created");

    let mut display = Builder::new(ST7789, di)
        .reset_pin(rst_display)
        .display_size(240, 320)
        .orientation(Orientation::new().rotate(Rotation::Deg90))
        .invert_colors(ColorInversion::Inverted)
        .init(&mut delay)
        .unwrap();
    log::info!("Display created");

    let start = time::now();
    display.clear(Rgb565::BLUE).unwrap();
    log::info!("clear drawing time {}", time::now() - start);

    delay.delay_millis(500);

    let fullscreen = Rectangle::new(Point::new(0, 0), Size::new(320, 240));
    let start = time::now();
    display.fill_solid(&fullscreen, Rgb565::RED).unwrap();
    log::info!("solid drawing time {}", time::now() - start);

    loop {
        delay.delay(500.millis());
    }
}
```

joyel24, 2025-01-26:
It's very slow with an ESP32-S3 also. 4 lines in a loop looks: https://github.com/almindor/mipidsi/issues/83#issuecomment-2614190771

georgik, 2025-01-27:
Any ideas about the DMA problem @bjoernQ?

Dominaezzz, 2025-02-13:
If you're experiencing issues with DMA on `esp-hal`, feel free to open an issue/discussion there and I (or the maintainers) can take a look.
I'm particularly interested in improving the performance of DMA on esp32s and making sure every use case can be satisfied by the esp-hal APIs, so I'd appreciate if you did.

Be-ing, 2025-04-01:
> We can use #[maybe_async] to keep most code singular, but the same issue is hit with embedded-graphics

I strongly recommend using the [bisync crate](https://docs.rs/bisync/latest/bisync/) instead of maybe_async. I've used bisync to add async support to [embedded-sdmmc](https://github.com/rust-embedded-community/embedded-sdmmc-rs/pull/176) and [apa102-spi](https://github.com/smart-leds-rs/apa102-spi-rs/pull/17) and I'm quite happy with how it works; as far as I am aware, it's the most maintainable way for libraries to provide both blocking and async APIs.

maybe_async has two huge flaws. First, its behavior changes based on Cargo features *on the macro crate* rather than using the macro to generate feature-gated code in the crate that the macro is invoked in. Because Cargo features are additive, separate crates in the dependency graph using maybe_async [can break each other](https://github.com/fMeow/maybe-async-rs/issues/21). The other issue is that a crate using maybe_async can't be built simultaneously for synchronous and asynchronous usage; they're mutually exclusive because maybe_async uses the presence/absence of a single Cargo feature rather than separate Cargo features for sync and async.

georgik, 2025-04-01:
BTW: Here's new example that is using DMA with esp-hal 1.0.0 and Bevy ECS - https://github.com/georgik/esp32-conways-game-of-life-rs

Just I would recommend to be careful about SPI frequency. Most displays are operating at 40MHz. Cranking it up to 80 might work, just double check the datasheet. I know that 60 MHz was causing many weird problems during initializations.

davide-scalzo, 2025-06-28:
Thank you @almindor , I change the hot loop like so:

```rust
loop {
        let temp = temp_sensor.get_temperature();
        let temp_c = temp.to_celsius() - 17.0f32;
        let arc_size = 128;
        let sweep = temp_c * 360.0 / 60.0;

        let center = display.bounding_box().center();
        if target_rgb == Rgb565::CSS_DARK_SLATE_GRAY {
            target_rgb = Rgb565::RED;
        } else {
            target_rgb = Rgb565::CSS_DARK_SLATE_GRAY;
        }

        let now = Instant::now();
        let _ = display.clear(target_rgb);
        let end = Instant::now();
        let elapsed = end - now;
        info!("Clearing display took: {:?}", elapsed.as_millis());

        let now = Instant::now();
        Arc::new(
            Point::new(center.x - arc_size / 2, center.y - arc_size / 2),
            128 - 4,
            90.0.deg(),
            360.0.deg(),
        )
        .into_styled(arc_stroke_bg)
        .draw(&mut display)
        .ok();

        Arc::new(
            Point::new(center.x - arc_size / 2, center.y - arc_size / 2),
            128 - 4,
            90.0.deg(),
            sweep.deg(),
        )
        .into_styled(arc_stroke)
        .draw(&mut display)
        .ok();

        // Draw the text
        Text::with_text_style(
            format!("{:.1}°C", temp_c).as_str(),
            center,
            char_style,
            text_style,
        )
        .draw(&mut display)
        .expect("Unable to write text");

        let end = Instant::now();
        let elapsed = end - now;
        info!("Drawing stuff took: {:?}", elapsed.as_millis());

        Timer::after(Duration::from_secs(1)).await;
    }
```

Changing display clearing colour each iteration does not improve performance, but splitting the timings tracking, shows that it is somewhat faster than drawing the circles.

[image]

almindor, 2025-07-04:
> Thank you [@almindor](https://github.com/almindor) , I change the hot loop like so:
> 
> loop {
>         let temp = temp_sensor.get_temperature();
>         let temp_c = temp.to_celsius() - 17.0f32;
>         let arc_size = 128;
>         let sweep = temp_c * 360.0 / 60.0;
> 
>         let center = display.bounding_box().center();
>         if target_rgb == Rgb565::CSS_DARK_SLATE_GRAY {
>             target_rgb = Rgb565::RED;
>         } else {
>             target_rgb = Rgb565::CSS_DARK_SLATE_GRAY;
>         }
> 
>         let now = Instant::now();
>         let _ = display.clear(target_rgb);
>         let end = Instant::now();
>         let elapsed = end - now;
>         info!("Clearing display took: {:?}", elapsed.as_millis());
> 
>         let now = Instant::now();
>         Arc::new(
>             Point::new(center.x - arc_size / 2, center.y - arc_size / 2),
>             128 - 4,
>             90.0.deg(),
>             360.0.deg(),
>         )
>         .into_styled(arc_stroke_bg)
>         .draw(&mut display)
>         .ok();
> 
>         Arc::new(
>             Point::new(center.x - arc_size / 2, center.y - arc_size / 2),
>             128 - 4,
>             90.0.deg(),
>             sweep.deg(),
>         )
>         .into_styled(arc_stroke)
>         .draw(&mut display)
>         .ok();
> 
>         // Draw the text
>         Text::with_text_style(
>             format!("{:.1}°C", temp_c).as_str(),
>             center,
>             char_style,
>             text_style,
>         )
>         .draw(&mut display)
>         .expect("Unable to write text");
> 
>         let end = Instant::now();
>         let elapsed = end - now;
>         info!("Drawing stuff took: {:?}", elapsed.as_millis());
> 
>         Timer::after(Duration::from_secs(1)).await;
>     }
> Changing display clearing colour each iteration does not improve performance, but splitting the timings tracking, shows that it is somewhat faster than drawing the circles. Still... extremely slow :(
> 
> <img alt="Image" width="294" src="https://private-user-images.githubusercontent.com/749029/460248778-d3e0d1ef-e4d1-410b-8e80-ac98c71ab2cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTE2NDYxMzgsIm5iZiI6MTc1MTY0NTgzOCwicGF0aCI6Ii83NDkwMjkvNDYwMjQ4Nzc4LWQzZTBkMWVmLWU0ZDEtNDEwYi04ZTgwLWFjOThjNzFhYjJjZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcwNFQxNjE3MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yODA3YTlhOTI2ZTIzNTBjMjQxYmEyMGQzMmY3YzY0YWQzZWM1Mjc3NzM3MTM0MGVkM2UyZGFhZTQ4OGRlNjI2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.osnVpVDBJyUSTdszKQNlridTW5-tCvgYoSwpKoE-8a0">

I think this looks feasible. The clears are 24ms, if we take things into bits my napkin math tells me:
`(172*320*3*8) / 40000000 = ~22ms` so doing a clear in 24 is expected since it has to do a few more things around each transfer. 

Not sure if there is anything that can be done better here. The letters/shapes are probably optimizable better given embedded-graphgics and mipidsi both, but that's a bigger pull.

oletf, 2026-09-07:
Hi, 

I've noticed some big difference between the expected theoretical data transfer time and what I get :
using an ESP32-C6 with SpiDma @ 80MHz, 480×320 screen with Rgb565

I've measured the following :
```rust
  let mut fb =
        Framebuffer::<Rgb565, _, LittleEndian, W, H, { buffer_size::<Rgb565>(W, H) }>::new();

  // * 31ms
  let start_clear = Instant::now();
  let _ = display.clear(Rgb565::WHITE);
  debug!("clear: {=u64:tus}", start_clear.elapsed().as_micros());

  // * 41ms
  let start_set_pixels = Instant::now();
  let _ = display.set_pixels(
      0,
      0,
      (W - 1) as _,
      (H - 1) as _,
      fb.data()
          .as_chunks::<{ size_of::<Rgb565>() }>()
          .0
          .iter()
          .map(|&c| Rgb565::from(RawU16::from(u16::from_le_bytes(c)))),
  );
  debug!(
      "set_pixels: {=u64:tus}",
      start_set_pixels.elapsed().as_micros()
  );

  // * 49ms
  let start_draw_img = Instant::now();
  let _ = fb.as_image().draw(&mut display);
  debug!("draw_img: {=u64:tus}", start_draw_img.elapsed().as_micros());
```
here's the expected optimal transfer time:
```
 480 × 320 × 16 bit / 80 megabps = 30.72 ms
```
- 31ms for the clear : makes sense to me (and hints me that Dma and Spi speed are set properly)
- 41ms for the manual `set_pixels` with manual bytes to color conversion of the FrameBuffer : 
   kinda double bytes>color>bytes conversion with [`InterfacePixelFormat::send_pixels`](https://github.com/almindor/mipidsi/blob/c40e225973c090c78e0c52f483fa0811e7d07b3f/src/interface.rs#L106-L111) called inside, 
  but the overhead seems decent : still above 24fps
- 49ms for the last one, which seemed like the most intuitive to write to me:
    which somehow gets an extra 8ms, which I can't figure out the reason for...
    [`ImageRaw::draw`](https://github.com/embedded-graphics/embedded-graphics/blob/59fd64be61167682e9b3fe8e73d02c431c87b25a/src/image/image_raw.rs#L213-L216) calls the same `fill_continuous`, and the only notable difference from the previous case I notice, is the `ContiguousPixels` iterator.

Would there be a better way to flush a FrameBuffer to Spi at full speed ? 
I also noticed this [`set_pixels_from_buffer`](https://github.com/almindor/mipidsi/pull/143) function which existed in the past, and seems to be exactly what I'm after, but I couldn't find the reason why it got removed, just that https://github.com/almindor/mipidsi/pull/149 had it removed.

Thanks forward !