almindor/mipidsi #184: Using rp2040 with st7789

kpcyrd, 2025-08-28:
**Describe the bug**

I'm getting the following build error when trying to use this crate together with rp2040 SPI:

```
error[E0277]: the trait bound `Spi<Enabled, SPI1, (Pin<Gpio15, FunctionSpi, PullDown>, Pin<Gpio14, FunctionSpi, PullDown>)>: SpiDevice` is not satisfied
   --> src/main.rs:68:40
    |
68  |     let di = SpiInterface::<_, _>::new(spi, dc, &mut buffer);
    |              ------------------------- ^^^ the trait `embedded_hal::spi::SpiDevice` is not implemented for `Spi<Enabled, SPI1, (Pin<Gpio15, FunctionSpi, PullDown>, Pin<Gpio14, FunctionSpi, PullDown>)>`
    |              |
    |              required by a bound introduced by this call
    |
note: there are multiple different versions of crate `embedded_hal` in the dependency graph
   --> /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embedded-hal-1.0.0/src/spi.rs:347:1
    |
347 | pub trait SpiDevice<Word: Copy + 'static = u8>: ErrorType {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ this is the required trait
    |
   ::: src/main.rs:8:5
    |
8   | use waveshare_rp2040_zero::{
    |     --------------------- one version of crate `embedded_hal` used here, as a dependency of crate `cortex_m`
...
63  |         embedded_hal::spi::MODE_0,
    |         ------------ one version of crate `embedded_hal` used here, as a direct dependency of the current crate
    |
   ::: /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/rp2040-hal-0.10.2/src/spi.rs:178:1
    |
178 | pub struct Spi<S: State, D: SpiDevice, P: ValidSpiPinout<D>, const DS: u8 = 8u8> {
    | -------------------------------------------------------------------------------- this type doesn't implement the required trait
    |
   ::: /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embedded-hal-0.2.7/src/adc.rs:39:1
    |
39  | pub trait Channel<ADC> {
    | ---------------------- this is the found trait
    = help: you can use `cargo tree` to explore your dependency tree
note: required by a bound in `SpiInterface::<'a, SPI, DC>::new`
   --> /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/mipidsi-0.9.0/src/interface/spi.rs:28:15
    |
28  | impl<'a, SPI: SpiDevice, DC: OutputPin> SpiInterface<'a, SPI, DC> {
    |               ^^^^^^^^^ required by this bound in `SpiInterface::<'a, SPI, DC>::new`
29  |     /// Create new interface
30  |     pub fn new(spi: SPI, dc: DC, buffer: &'a mut [u8]) -> Self {
    |            --- required by a bound in this associated function
    = note: the full name for the type has been written to '/home/user/repos/kpcyrd/rp2040-demo-st7789/target/thumbv6m-none-eabi/debug/deps/rp2040_demo_st7789-2952a7fd830702c2.long-type-1277235311542819856.txt'
    = note: consider using `--verbose` to print the full type name to the console
```

I'm slightly confused because `SpiDevice` seems to be the trait implemented by `pac.SPI1`.

**To Reproduce**

I'm using the following code:

```rust
#![no_std]
#![no_main]

use defmt_rtt as _;
use mipidsi::{interface::SpiInterface, models::ST7789};
use panic_halt as _;
use waveshare_rp2040_zero::{
    Pins, XOSC_CRYSTAL_FREQ, entry,
    hal::{
        Sio,
        clocks::{Clock, init_clocks_and_plls},
        fugit::RateExtU32,
        gpio, pac,
        rosc::RingOscillator,
        spi::{self, SpiDevice},
        timer::Timer,
        watchdog::Watchdog,
    },
};

#[entry]
fn main() -> ! {
    let mut pac = pac::Peripherals::take().unwrap();

    // Configure clocks and timers
    let mut watchdog = Watchdog::new(pac.WATCHDOG);
    let clocks = init_clocks_and_plls(
        XOSC_CRYSTAL_FREQ,
        pac.XOSC,
        pac.CLOCKS,
        pac.PLL_SYS,
        pac.PLL_USB,
        &mut pac.RESETS,
        &mut watchdog,
    )
    .ok()
    .unwrap();

    // Configure gpio
    let sio = Sio::new(pac.SIO);
    let pins = Pins::new(
        pac.IO_BANK0,
        pac.PADS_BANK0,
        sio.gpio_bank0,
        &mut pac.RESETS,
    );

    // Set up our SPI pins so they can be used by the SPI driver
    let mosi = pins.gp15.into_function::<gpio::FunctionSpi>();
    let sck = pins.gp14.into_function::<gpio::FunctionSpi>();
    let cs = pins.gp9.into_push_pull_output();
    let dc = pins.gp12.into_push_pull_output();
    // let reset = pins.gp11.into_push_pull_output();

    let spi = spi::Spi::<_, _, _, 8>::new(pac.SPI1, (mosi, sck));

    // Exchange the uninitialised SPI driver for an initialised one
    let mut spi = spi.init(
        &mut pac.RESETS,
        clocks.peripheral_clock.freq(),
        16_u32.MHz(),
        embedded_hal::spi::MODE_0,
    );

    // Configure display
    let mut buffer = [0_u8; 512];
    let di = SpiInterface::<_, _>::new(spi, dc, &mut buffer);

    loop {}
}
```

**Cargo.toml**

<details>

```toml
[package]
name = "rp2040-demo-st7789"
version = "0.1.0"
edition = "2024"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

# Profiles taken from https://github.com/rp-rs/rp2040-project-template

[profile.dev]
codegen-units = 1
debug = 2
debug-assertions = true
incremental = false
opt-level = 3
overflow-checks = true

[profile.release]
codegen-units = 1
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 3
overflow-checks = false

[dependencies]
cortex-m-rt = "0.7.3"
defmt-rtt = "1"
eh0 = { package = "embedded-hal", version = "0.2" }
embedded-graphics = "0.8.1"
embedded-hal = "1"
embedded-hal-compat = "0.13.0"
#fugit = "0.3.7"
mipidsi = "0.9.0"
#nb = "1.1.0"
panic-halt = "1"
waveshare-rp2040-zero = "0.8"
```

</details>

**Expected behavior**

Code compiles

**Used hardware and software**
Please provide details about the hardware and software you are using:
- `mipidsi` version: `0.9.0`
- MCU: rp2040
- HAL: rp2040-hal 0.10.0
- Display Controller IC: ST7789
- Type of connection: e.g. SPI

SpamixOfficial, 2025-09-27:
Hey! 

Sorry for reviving this thread but I am having some issues like this with embassy. I am using embassy (rev `9158bfde62fce3de4cfb0478e40f8f2b0d5a2ceb`) and I am trying to use a display over SPI. Because I have multiple peripherals over SPI I have to use a shared bus, but trying to use the shared bus for the SpiInterface is giving me the following compiler error:
```
error[E0277]: the trait bound `SpiDevice<'_, NoopRawMutex, Spi<'static, SPI0, Async>, ...>: SpiDevice` is not satisfied
   --> src/tasks.rs:17:40
    |
 17 |     let di = SpiInterface::<_, _>::new(dev, dcx, &mut buffer);
    |              ------------------------- ^^^ the trait `embedded_hal::spi::SpiDevice` is not implemented for `SpiDevice<'_, NoopRawMutex, Spi<'static, SPI0, Async>, Output<'_>>`
    |              |
    |              required by a bound introduced by this call
    |
note: there are multiple different versions of crate `embedded_hal` in the dependency graph
   --> /Users/alexander/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embedded-hal-1.0.0/src/spi.rs:347:1
    |
347 | pub trait SpiDevice<Word: Copy + 'static = u8>: ErrorType {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ this is the required trait
    |
   ::: src/controls.rs:1:5
    |
  1 | use embassy_rp::{
    |     ----------
    |     |
    |     one version of crate `embedded_hal` used here, as a dependency of crate `embassy_rp`
    |     one version of crate `embedded_hal` used here, as a dependency of crate `cortex_m`
    |
   ::: /Users/alexander/.cargo/git/checkouts/embassy-c08a80187403f815/9158bfd/embassy-embedded-hal/src/shared_bus/asynch/spi.rs:39:1
    |
 39 | pub struct SpiDevice<'a, M: RawMutex, BUS, CS> {
    | ---------------------------------------------- this type doesn't implement the required trait
    |
   ::: /Users/alexander/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embedded-hal-0.2.7/src/adc.rs:39:1
    |
 39 | pub trait Channel<ADC> {
    | ---------------------- this is the found trait
    = help: you can use `cargo tree` to explore your dependency tree
note: required by a bound in `SpiInterface::<'a, SPI, DC>::new`
   --> /Users/alexander/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/mipidsi-0.9.0/src/interface/spi.rs:28:15
    |
 28 | impl<'a, SPI: SpiDevice, DC: OutputPin> SpiInterface<'a, SPI, DC> {
    |               ^^^^^^^^^ required by this bound in `SpiInterface::<'a, SPI, DC>::new`
 29 |     /// Create new interface
 30 |     pub fn new(spi: SPI, dc: DC, buffer: &'a mut [u8]) -> Self {
    |            --- required by a bound in this associated function
    = note: the full name for the type has been written to '/Users/alexander/Documents/git/spamixofficial/schedule_viewer_rp2040/target/thumbv6m-none-eabi/debug/deps/schedule_viewer_rp2040-7e1f2ceefa815b08.long-type-2909129660130174570.txt'
    = note: consider using `--verbose` to print the full type name to the console

```

Here is my current code (excuse the half-assed variable names!):
```rust
use embassy_embedded_hal::shared_bus::asynch::spi::SpiDevice;
use embassy_rp::gpio::Output;
use embassy_rp::peripherals::SPI0;
use embassy_rp::spi::{self, Spi};
use embassy_sync::blocking_mutex::raw::NoopRawMutex;
use embassy_sync::mutex::Mutex;
use mipidsi::interface::SpiInterface;
use mipidsi::models::ST7735s;
use mipidsi::Builder;

type SpiBus = Mutex<NoopRawMutex, Spi<'static, SPI0, spi::Async>>;

#[embassy_executor::task]
async fn screen_painter(bus: &'static SpiBus, cs: Output<'static>, bl: Output<'static>, dcx: Output<'static>, rst: Output<'static>) {
    let dev = SpiDevice::new(&bus, cs);
    let mut buffer = [0_u8; 512];
    let di = SpiInterface::<_, _>::new(dev, dcx, &mut buffer); // Errors here

    let mut disp = Builder::new(ST7735s, di);
}
```

I've been trying to solve this for hours at this point, would appreciate any help!!

SpamixOfficial, 2025-09-27:
FYI: I solved it, was mixing blocking and async embedded_hal by accident