almindor/mipidsi #187: Blank Screen with ST7789 on ESP32-S3

twatzl, 2025-09-21:
Sorry. Have to bother you with another one of those blank screen issues. But I have tried everything and still cannot even get a hint of what might be wrong.

I am trying to run an ESP32-S3 using esp_hal, but so far I had no luck getting the screen to work.

I started with the ESP32-C3 sample and tried upgrading it to latest esp-hal (1.0.0-rc.0).

cargo.toml
```toml
[package]
name = "spi_ili9486_esp32s3"
version = "0.1.0"
edition = "2021"

[build]
target = "xtensa-esp32s3-none-elf"

[dependencies]
mipidsi = { path = "../../" }
hal = { package = "esp-hal", version = "1.0.0-rc.0", features = ["esp32s3", "unstable", "rt"] }
esp-backtrace = { version = "0.17", features = [
    "esp32s3",
    "panic-handler",
    "exception-handler",
    "println",
] }
esp-println = { version = "0.15", features = ["esp32s3"] }
embedded-graphics = "0.8.1"
fugit = "0.3.7"
embedded-hal = "1.0.0"
embedded-hal-bus = "0.3.0"

[profile.dev.package.esp-hal]
debug = false

[workspace]
```

main.rs
```rust
#![no_std]
#![no_main]

use esp_println::println;

use embedded_hal::delay::DelayNs;
use embedded_hal_bus::spi::ExclusiveDevice;
/* --- Needed by ESP32-s3 --- */
use esp_backtrace as _;
use hal::{
    main,
    delay::Delay,
    gpio::{Level, Output},
    rtc_cntl::Rtc,
    spi::{master::Spi, master::Config, Mode},
    timer::timg::TimerGroup,
    time::Rate
};
/* -------------------------- */

use embedded_graphics::{
    pixelcolor::Rgb565,
    prelude::*,
    primitives::{Circle, Primitive, PrimitiveStyle, Triangle},
};

// Provides the parallel port and display interface builders
use mipidsi::interface::SpiInterface;

// Provides the Display builder
use mipidsi::{models::ST7789, Builder};

use fugit::RateExtU32;
use hal::gpio::OutputConfig;

// Display
const W: i32 = 170;
const H: i32 = 320;

#[main]
fn main() -> ! {
    let peripherals = hal::init(hal::Config::default());
    // let peripherals = Peripherals::take();
    // let system = SystemControl::new(peripherals.SYSTEM);
    // let clocks = ClockControl::boot_defaults(system.clock_control).freeze();
    let mut delay = Delay::new();

    // Disable the RTC and TIMG watchdog timers
    let mut rtc = Rtc::new(peripherals.LPWR);
    let timer_group0 = TimerGroup::new(peripherals.TIMG0);
    let mut wdt0 = timer_group0.wdt;
    let timer_group1 = TimerGroup::new(peripherals.TIMG1);
    let mut wdt1 = timer_group1.wdt;
    rtc.swd.disable();
    rtc.rwdt.disable();
    wdt0.disable();
    wdt1.disable();

    // Define the delay struct, needed for the display driver
    let mut delay = Delay::new();

    let dc_pin = peripherals.GPIO7;
    let rst_pin = peripherals.GPIO5;
    let led_pin = peripherals.GPIO38;

    // spi pins
    let sck = peripherals.GPIO12;
    let miso = peripherals.GPIO11;
    let mosi = peripherals.GPIO13;
    let cs = peripherals.GPIO10;

    // Define the Data/Command select pin as a digital output
    let dc = Output::new(dc_pin, Level::Low, OutputConfig::default());
    // Define the reset pin as digital outputs and make it high
    let mut rst = Output::new(rst_pin, Level::High, OutputConfig::default());
    let _led = Output::new(led_pin, Level::High, OutputConfig::default());

    // Define the SPI pins and create the SPI interface
    let spi = Spi::new(peripherals.SPI2, Config::default()
        .with_frequency(Rate::from_mhz(1))
        .with_mode(Mode::_0))
    .unwrap()
    .with_sck(sck)
    .with_mosi(mosi)
    .with_miso(miso)
    .with_cs(hal::gpio::NoPin);

    let cs_output = Output::new(cs, Level::High, OutputConfig::default());
    let spi_device = ExclusiveDevice::new_no_delay(spi, cs_output).unwrap();

    // Define the display interface with no chip select
    let mut buffer = [0_u8; 512];
    let di = SpiInterface::new(spi_device, dc, &mut buffer);

    // Define the display from the display interface and initialize it
    let mut display = Builder::new(ST7789, di)
        .display_size(W as u16, H as u16)
        .reset_pin(rst)
        .init(&mut delay)
        .unwrap();

    delay.delay_ms(1000);

    // Make the display all black
    display.clear(Rgb565::BLUE)
        .map_err(|err| println!("Error clearing display: {:?}", err));
    println!("Display cleared!");

    // Draw a smiley face with white eyes and a red mouth
    draw_smiley(&mut display).unwrap();

    loop {
        // Do nothing
    }
}

fn draw_smiley<T: DrawTarget<Color = Rgb565>>(display: &mut T) -> Result<(), T::Error> {
    // Draw the left eye as a circle located at (50, 100), with a diameter of 40, filled with white
    Circle::new(Point::new(50, 100), 40)
        .into_styled(PrimitiveStyle::with_fill(Rgb565::WHITE))
        .draw(display)?;

    // Draw the right eye as a circle located at (50, 200), with a diameter of 40, filled with white
    Circle::new(Point::new(50, 200), 40)
        .into_styled(PrimitiveStyle::with_fill(Rgb565::WHITE))
        .draw(display)?;

    // Draw an upside down red triangle to represent a smiling mouth
    Triangle::new(
        Point::new(130, 140),
        Point::new(130, 200),
        Point::new(160, 170),
    )
    .into_styled(PrimitiveStyle::with_fill(Rgb565::RED))
    .draw(display)?;

    // Cover the top part of the mouth with a black triangle so it looks closed instead of open
    Triangle::new(
        Point::new(130, 150),
        Point::new(130, 190),
        Point::new(150, 170),
    )
    .into_styled(PrimitiveStyle::with_fill(Rgb565::BLACK))
    .draw(display)?;

    Ok(())
}
```

The code seems to run all fine, no panic or anything, just the screen is blank. Maybe it would be good to add a debug guide for issues like these? I have seen quite a lot popped up.

What I have tried
- reduce frequency
- change spi mode
- check pin layout with the existing arduino driver (which is confirmed working on the exact screen)... 3 times..

rfuest, 2025-09-22:
I tried your code and it works for my display. The only change to the Rust code I made was to use other GPIO pins for some signals, because the only ESP32 S3 board I have available doesn't expose the pins you used:
```rust
let rst_pin = peripherals.GPIO1;
let dc_pin = peripherals.GPIO2;
let led_pin = peripherals.GPIO3;
```

Does your display have the `MISO` line connected? It shouldn't matter, because `mipidsi` currently doesn't use it, but my display module doesn't have a `MISO` pin and I left `GPIO11` floating.

Why are you using a local copy of `mipidsi`? I did use version `0.9.0` for my test.

Using other GPIO pins shouldn't make a difference, but I would suggest that you try to use the same ones I used (if they are accessible). If that doesn't change anything please upload the entire project somewhere. The code didn't compile without adding a `.cargo/config.toml` and perhaps there is some difference in the build settings I used.

> Maybe it would be good to add a debug guide for issues like these? I have seen quite a lot popped up.

It would be a good idea to extend the troubleshooting guide, but I'm not sure what additional information would be useful. There are many little details that can cause the display to not work.

twatzl, 2025-09-30:
OH MY GOD. I have been trying to get this to run for 2 months. And now.. Finally it is working. I am so happy.  😅

Adding a section about parallel interfaces and when you need what in the docs would definitely be VERY helpful.

Thanks for all your help!